openwrt/luci #1291: [luci-app-openvpn] Push and other config options quickly disappear after Advanced Configuration->VPN is entered

Menion2k, 2017-08-09:
Hi. I'm running Luci openvpn-app 17.219.28675-9ee26ac-1 in LEDE
Also with the previous version, if I enter the VPN tab in Advanced Configuration, several usefull push parameters are quickly displayed, then disappear leaving only "Configure cliend mode"
Attached a screenshot I have quickly taken when entering in the VPN tab:

[image]

This is the UCI config file content:

```
config openvpn 'myvpn'
        option verb '3'
        option proto 'tcp'
        option dev 'tun'
        option log '/tmp/openvpn.log'
        option ca '/etc/openvpn/ca.crt'
        option cert '/etc/openvpn/my-server.crt'
        option key '/etc/openvpn/my-server.key'
        option dh '/etc/openvpn/dh2048.pem'
        list push 'redirect-gateway def1'
        list push 'dhcp-option DNS 192.168.182.1'
        option port '8294'
        option server '10.12.0.0 255.255.255.0'
        option keepalive '10 120'
        option enabled '1'
        option comp_lzo 'adaptive'

```

Wedmer, 2017-08-11:
Looks like duplicate #887 
Can you provide screenshot after options disappeared?

Menion2k, 2017-08-11:
Yes, I confirm it is the duplicate of #887
Here you have the "after" screenshot:

[image]

Wedmer, 2017-08-11:
So you have to get upstream version then.
This problem was fixed with #1123.

Menion2k, 2017-08-11:
Thanks
I run LEDE, so I guess I have to manually patch or wait for version bump

Menion2k, 2017-08-14:
Hi
Actually it seems that the lede-17.01 is not updated
Who is in charge of align it with master?

adilou1981, 2018-08-02:
Hello,

I am using git-18.201.27126-7bf0367-1  and still get the problem ? !

gigafunk, 2018-09-06:
Powered by LuCI openwrt-18.06 branch (git-18.228.31946-f64b152) / OpenWrt 18.06.1 r7258-5eb055306f 

I deleted the sample configs before I tried to make a new one and now I have no options at all
[image]

dibdot, 2018-10-23:
shoul be fixed in trunk.

closed 2018-10-23 by dibdot