openwrt/luci #1300: luci-mod-rpc not working anymore

dh-harald, 2017-08-12:
It looks, I rpc stops working. After a successful login, I can't use the token anymore...
Request:
```
POST /cgi-bin/luci/rpc/auth HTTP/1.1
Host: xxx.xxx.xxx.xxx
Connection: keep-alive
Accept-Encoding: gzip, deflate
Accept: */*
User-Agent: python-requests/2.10.0
Content-Length: 51

{"params": ["root", "xxxxxxxxx"], "method": "login"}
```
Response:
```
HTTP/1.1 200 OK
Connection: close
Transfer-Encoding: chunked
Content-Type: application/json
Set-Cookie: sysauth=e22f633c44a1e06152aac3f7bf3285d1; path=/
Cache-Control: no-cache
Expires: 0

44
{"id":null,"result":"e22f633c44a1e06152aac3f7bf3285d1","error":null}
```

Request:
```
POST /cgi-bin/luci/rpc/sys?auth=e22f633c44a1e06152aac3f7bf3285d1 HTTP/1.1
Host: xxx.xxx.xxx.xxx
Connection: keep-alive
Accept-Encoding: gzip, deflate
Accept: */*
User-Agent: python-requests/2.10.0
Content-Length: 40

{"params": [], "method": "net.arptable"}
```

Response:
```
HTTP/1.1 403 Forbidden
Connection: close
Transfer-Encoding: chunked
```

I've tried the following without any help
- using existing token (logged in via web and grab the token)
- using cookie sysauth instead of using auth=xxx parameter

I'm compiling lede weekly basis, currently, I'm using:
luci-mod-rpc - git-17.223.47793-8bc1971-1

This problem exists about a month ago.

dh-harald, 2017-08-15:
I've dig a little bit into this issue...
The problematic commit was:
https://github.com/openwrt/luci/commit/4e1b884157017b6729d2523e95385756898eac7c

I've find the problem here:
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/dispatcher.lua#L178

sdat.values.username should be sdat.values.user
This modification solves the rpc login problem

@jow- : I didn't raise a PR, just because I'm not sure about the consequences the following lines:
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/dispatcher.lua#L189
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/dispatcher.lua#L420

bojanpog, 2017-09-05:
I have the same problem. 

Openwrt daily snapshot around two month skips with this bug.

Is there eny hope for solving this?

I've tried to change dispatcher.lua in openwrt but it is just like that this source has to be compiled first to take effect?

dh-harald, 2017-09-05:
i'm using this patch to recompile luci:
`diff --git a/modules/luci-base/luasrc/dispatcher.lua b/modules/luci-base/luasrc/dispatcher.lua
index e4f77f1..386c036 100644
--- a/modules/luci-base/luasrc/dispatcher.lua
+++ b/modules/luci-base/luasrc/dispatcher.lua
@@ -175,7 +175,8 @@ local function session_retrieve(sid, allowed_users)
 	   type(sdat.values) == "table" and
 	   type(sdat.values.token) == "string" and
 	   (not allowed_users or
-	    util.contains(allowed_users, sdat.values.username))
+	    util.contains(allowed_users, sdat.values.username) or
+	    util.contains(allowed_users, sdat.values.user))
 	then
 		return sid, sdat.values
 	end
`
It's an ugly hack, but at least, it's working

bojanpog, 2017-09-05:
Very clever !

It is working, thank you! 🥇

xinghusp, 2017-10-22:
The same problem here...

stokito, 2018-03-08:
I tested today's snapshot and the problem is still persists. @dh-harald it would be great if you create a pull request with the fix.
Thank you in advance

h3rj4n, 2018-04-04:
Just updated to the latest version of OpenWRT. Cost me a few hours to figure this out. Is it easy to apply the patch and rebuild luci?

closed 2018-04-06 by jow-

dh-harald, 2018-04-14:
jow-: it doesn't fix this issue. It works with cookie method, but it's not working with auth GET parameter :(

bojanpog, 2018-05-04:
Hello!

jsonrpc is not working any more again with auth GET parameter

dh-harald pasted pa patch 6 month ago at https://github.com/openwrt/luci/issues/1300 , but it doesn work any more.. 

Again getting error 403

@dh-harald: did you figured out, what is the problem with GET auth this time?

Regards

commit feefc600 (2018-04-06): luci-mod-rpc: rework authentication and session handling

 - Use the ubus session.login procedure to authenticate credentials
 - Fix testing of allowed usernames
 - Support authentication via sysauth cookie

Fixes #1300, #1700, #1711