openwrt/luci #1312: Can any one explain me this?

kuoruan, 2017-08-18:
https://github.com/openwrt/luci/blob/master/modules/luci-base/luasrc/sys.lua#L654
```
(init_action("enable", name) == 1)
```
Why there is 1, but not 0?

https://htmlpreview.github.io/?https://raw.githubusercontent.com/openwrt/luci/master/documentation/api/modules/luci.sys.init.html#init.enable

https://github.com/lede-project/source/blob/master/package/base-files/files/etc/rc.common#L48

dibdot, 2017-08-18:
successful 'enable' execution comes back with '1', e.g.:
 
```
root@blackhole:~# /etc/init.d/adblock enable; echo $?
1

```

kuoruan, 2017-08-19:
@dibdot But in ```/etc/rc.common```, ```enable``` is defined return 1 when there has no ```START``` and ```STOP```. It means enable failed.

dibdot, 2017-08-19:
Usually 'START' and/or 'STOP' variables defined in every init.d script at the first place. As long as you do not have defined both variables, this code block comes always back with rc '1', let's take adblock init with 'enable' action as example again:

Script source (just added set -x and two debug lines in rc.common):
```
#!/bin/sh /etc/rc.common
#
set -x
START=50
USE_PROCD=1
[...]
```
Script execution output (only the relevant section):
```
+ [ -n 50 -o -n  ]
+ [ 50 ]
+ ln -s ../init.d/adblock /etc/rc.d/S50adblock
+ echo START-DEBUG/RC: 0
START-DEBUG/RC: 0
+ [  ]
+ echo STOP-DEBUG/RC: 1
STOP-DEBUG/RC: 1
```

Re-run with START **and** STOP variable ...
```
#!/bin/sh /etc/rc.common
#
set -x
START=50
STOP=10
USE_PROCD=1
[...]
```

Script execution output (only the relevant section):
```
+ [ -n 50 -o -n 10 ]
+ [ 50 ]
+ ln -s ../init.d/adblock /etc/rc.d/S50adblock
+ echo START-DEBUG/RC: 0
START-DEBUG/RC: 0
+ [ 10 ]
+ ln -s ../init.d/adblock /etc/rc.d/K10adblock
+ echo STOP-DEBUG/RC: 0
STOP-DEBUG/RC: 0
```

closed 2017-09-14 by kuoruan