openwrt/luci #1324: No Associated Stations displayed when using dynamic VLANs

Mattyx, 2017-08-22:
Hello,
I'm using SSIDs with WPA2 802.1X and dynamic VLANs.
For those SSIDs the "Associated Stations" section does not show any station.

I use 3 VLANs (with ID 18, 19 and 20).

I think this is due to the fact that these stations are associated to different devices than wlan0 or wlan1.
Using iw (or iwinfo) command I can see stations under the wlan0.20, wlan1.20, etc. devices. The parent device (wlan0 or wlan1) on this specific configuration does not have any station directly associated.

Maybe LuCI doesn't aggregate the additional VLAN interfaces under the "parent" interface?

There is a way to correct this?

Thanks in advance!

xsokolikx, 2019-04-29:
I'm very interested on this. Any solution to show the associated clients on a wireless dynamic vlan?
Thanks

ASoTNetworks, 2021-04-29:
I am also interested in seeing this fixed so we don't have to go into the command line to see connected clients on each VLAN interface.

gtxaspec, 2023-03-09:
2023 and still issue still persists.  clients connected to AP/VLAN do not show up on the associated stations list.

timfoley66, 2023-08-18:
Experiencing same issue.

spamtrap07, 2023-08-20:
I observe the same issue on openwrt 23.05.0-rc2.

hizukiayaka, 2024-11-03:
It is a history problem since the old openwrt time,
https://forum.archive.openwrt.org/viewtopic.php?id=71846

systemcrash, 2024-11-03:
I think the first reasonable steps to fix this including augmenting (read: patch) the iw command to aggregate station info irrespective of interface, if it doesn't already. Armed with this it should be trivial to figure out their vlans and which ap to roughly display them correctly.

What kind of information does iw have available?

tobiaswaldvogel, 2025-01-02:
Actually the code seems to be VLAN aware already:

[https://github.com/openwrt/luci/blob/master/modules/luci-base/htdocs/luci-static/resources/network.js](url)

```
	getAssocList: function() {
		var tasks = [];
		var ifnames = [ this.getIfname() ].concat(this.getVlanIfnames());

		for (var i = 0; i < ifnames.length; i++)
			tasks.push(callIwinfoAssoclist(ifnames[i]));

		return Promise.all(tasks).then(function(values) {
			return Array.prototype.concat.apply([], values);
		});
	},
```

but the function getVlanIfnames does not retrieve the VLANs. I did a quick and dirty test with just concatenating my wlan vlan interfaces to ifnames and that works perfectly well, i.e.iwinfo is called for every interface and all clients show up.

I'm also adding the VLAN to the returned object in order to display the right vlan interface instead of base interface.

```
                return Promise.all(tasks).then(function(values) {
+                       /* Add interface name */
+                       for (var i = 0; i < values.length; i++)
+                               for (b = 0; b < values[i].length; b++)
+                                       values[i][b]["ifname"] = ifnames[i];
+
                        return Array.prototype.concat.apply([], values);
                });
```

Once I figured out how to populate getVlanIfnames properly I will submit a patch.

tobiaswaldvogel, 2025-01-04:
So, at the end the data is coming from an ubus call to network.wireless status. You can check that on your device with ` ubus call network.wireless status` and there you will see that `vlans` is empty.

As far as I have seen this is answered by netifd and the notification to netifd should come from hostapd. I still need to dig a little bit further into this.

In the meanwhile I'm using a workaround in network.js to populate it, basically by comparing all interface with wireless extensions against what is reported from the `network.wireless`. The remaining ones are the vlan interfaces. They can also be matched against the radio device via the MAC, but if there are more than two APs on the same radio device this will not unique and and they are just added the to first AP. So this cannot be used as a final solution although for me this is good enough for now, as I don's have more than one SSID per radio device.

tobiaswaldvogel, 2025-01-05:
At the end it turns out that `dynamic_vlan` does not populate the network.wireless vlans array
So the best standard solutions is using wifi-vlan sections instead. It is a little bit more work unless you have a tremendous amount of VLANs, but anyway I prefer defining the bridges myself, as sometimes `wifi down` removed the br-lan bridge as well and I had to reboot the AP.

Nevertheless the vlan assignment based on the secret with `wpa_psk_file` still works the same way, which is probably what we are all after. So the setup is basically:

- 1 Create a proper switch configuration if applicable (you had that with dynamic vlan anyway)
- 2 Create a tagged interface for each VLAN via Luci
- 3 Create a bridge for each tagged interface
- 4 Create an interface for each bridge. If you don't need an IP you can leave it to proto none. In my setup the wifi APs and the router are separated, so the DHCP server for the VLANs runs on the router and not on the APs.
- 5 Create a wifi vlan entry for each VLAN in /etc/config/wireless. There is still no GUI option for this. E.g. 
```config wifi-vlan
        option vid '12'
        option network 'guest'
        option name 'gst'
```
`vid` is the vlanid in the wpa_psk_file, `network` is the interface name created in the previous step (covering the bridge), `name` is the suffix for the wifi vlan interface. If you want wifi vlans only for a specific wifi interface then you can also specify the option `iface` otherwise it will be created for all wifi interfaces

This setup provides the same functionality as the dynamic VLANs but with proper display in Luci, but with the base interface instead of the VLAN interface.
I'm going to submit a patch for Luci to address that.

spi43984, 2025-08-09:
I am using dynamic vlans both with eap as well as with ppsk. It works well on different platforms like Mikrotik and openwrt (but not so well on Unifi ;-).

Mikrotik has a nice feature to provide a comment for a user device via the radius attribute Mikrotik-Wireless-Comment. One can use it in Mikrotik's associated stations list to not only see a mac address but a device/user/whatsoever name as well.

I wonder if something like that could be implemented in the openwrt Associated Stations list as well.

damnms, 2026-08-19:
i have a lot of vlan's, managed by a freeradius server - nearly for every device (with different trust level). its a pity that dynamic vlans are not shown by default.