openwrt/luci #1327: interface management does not work

psyborg55, 2017-08-23:
trying to delete WAN6 interface results in nothing. trying to delete WAN interface disconnects both WAN and WAN6 interfaces but does not delete them.
traffic counter counts IPv4 traffic in WAN6 statistics.

 LEDE Reboot SNAPSHOT r4721-0168ba2

jow-, 2017-08-23:
wan6 is usually a simple IP alias to wan, the traffic counter is reporting stats of the underlying netdev. It is impossible for LuCI to differentiate traffic there.

0xFEEDC0DE64, 2017-10-09:
I could solve this by changing the device of WAN6 to one of the wifi interfaces and deleting it afterwards... still strange but it works!

psyborg55, 2017-10-09:
i think that's not needed, this bug is chrome-specific, it works fine in firefox. maybe just some netifd script tweaking

jow-, 2018-10-20:
Assuming that this has been fixed now.

closed 2018-10-20 by jow-