openwrt/luci #1328: Problem with compile openwrt with luci package

Deoptim, 2017-08-24:
Without Luci - it compile without problem.
If i chose in menuconfig luci - i have main error at stage of compiling the openwrt:
(compile without the **root** rights)
`install -d -m0755 /bin
install -m0755 src/po2lmo /bin/po2lmo
install: cannot create regular file '/bin/po2lmo': Permission denied
...
and Error
`
If i compile with root rights - I have NO this error...

64-bit /Ubuntu / 2 Core

jow-, 2017-08-24:
Your OpenWrt source it too old for this version of LuCI. Either update your OpenWrt or use an older LuCI release.

jow-, 2017-08-24:
In partcular you need commit https://github.com/openwrt/openwrt/commit/7d7858f6f1b95f1f11e02bb0e6e386f4fa93bcf5

closed 2017-08-24 by jow-