openwrt/luci #1331: x86 r8169 kernel error if the r8169 work as lan 1000 the fowling error get on heavy traffic

pantgr, 2017-08-30:
Powered by LuCI lede-17.01 branch (git-17.152.82987-7f6fc16) / LEDE Reboot 17.01.2 r3435-65eec8bd5f 

[ 1137.020013] ------------[ cut here ]------------
[ 1137.022631] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:306 0xc15fadd1()
[ 1137.025299] NETDEV WATCHDOG: eth2 (r8169): transmit queue 0 timed out
[ 1137.027936] Modules linked in: ath9k_htc ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath pppox ppp_generic nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE ds2482 cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY wire w83793 via_velocity via_rhine tmp103 tmp102 slhc sis900 sht21 rtc_pt7c4338 rtc_ds1307 r8169 pcnet32 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache ne2k_pci ltc4151 lm95241 lm92 lm90 lm85 lm77 lm75 lm63 iptable_mangle iptable_filter ipt_ECN ip_tables ina2xx ina209 e100 crc_ccitt compat bmp085_i2c bmp085 at24 adt7475 8390 8139too sch_cake nf_conntrack act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_tbf sch_htb sch_hfsc sch_ingress tmp421 k10temp gpio_fan adt7410 adt7x10 i2c_tiny_usb i2c_gpio ad799x industrialio_triggered_buffer kfifo_buf industrialio i2c_algo_pcf i2c_algo_pca gpio_pcf857x gpio_pca953x i2c_mux_pca954x i2c_mux_pca9541 i2c_mux_gpio i2c_mux i2c_dev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables hwmon_vid gpio_mcp23s08 3c59x e1000 ifb regmap_spi regmap_mmio regmap_i2c regmap_core lzo_decompress lzo_compress natsemi button_hotplug tg3 ptp pps_core mii libphy
[ 1137.062142] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.4.71 #0
[ 1137.065366] Hardware name: To Be Filled By O.E.M. To Be Filled By O.E.M./775i65G., BIOS P3.30 12/01/2009
[ 1137.068741]  00000000 c120e5d9 f680df68 c1841338 c1086d49 00000009 00000000 fffffff8
[ 1137.072080]  f68d0000 00000258 c1086da6 00000009 f680df68 c184421c f680df80 c15fadd1
[ 1137.072132]  c1841338 00000132 c184421c f68d0000 f8d9297d 00000000 f68d0000 00000000
[ 1137.072132] Call Trace:
[ 1137.072132]  [<c120e5d9>] ? 0xc120e5d9
[ 1137.072132]  [<c1086d49>] ? 0xc1086d49
[ 1137.072132]  [<c1086da6>] ? 0xc1086da6
[ 1137.072132]  [<c15fadd1>] ? 0xc15fadd1
[ 1137.072132]  [<c15fac00>] ? 0xc15fac00
[ 1137.072132]  [<c10c6d5f>] ? 0xc10c6d5f
[ 1137.072132]  [<c10c6efd>] ? 0xc10c6efd
[ 1137.072132]  [<c10899ed>] ? 0xc10899ed
[ 1137.072132]  [<c1089940>] ? 0xc1089940
[ 1137.072132]  [<c1047e2c>] ? 0xc1047e2c
[ 1137.072132]  <IRQ>  [<c1089be5>] ? 0xc1089be5
[ 1137.072132]  [<c1073bb0>] ? 0xc1073bb0
[ 1137.072132]  [<c16b51a5>] ? 0xc16b51a5
[ 1137.072132]  [<c104d9a1>] ? 0xc104d9a1
[ 1137.072132]  [<c104ddb6>] ? 0xc104ddb6
[ 1137.072132]  [<c10b506a>] ? 0xc10b506a
[ 1137.072132]  [<c18d3a21>] ? 0xc18d3a21
[ 1137.132517] ---[ end trace 7cf35aaefae8522a ]---

h3298, 2017-09-02:
Fowling error  ;-)
https://en.wikipedia.org/wiki/Bird_strike#/media/File:IAF_UH-60_after_birds_strike_outside.jpg

feckert, 2019-09-19:
This is not related to LuCI

closed 2019-09-19 by feckert