openwrt/luci #1340: [luci-app-openvpn] Several OpenVPN parameters missing from GUI

h3298, 2017-09-02:
**1.**
```
resolv-retry infinite
mssfix 'n'
```

(in /**usr/share/openvpn/openvpn.options** these are listed as supported, but dont seem to be in the GUI)
____________
**2.**  There is no GUI control to **enable log** and review it for connection errors (**option log '/etc/openvpn/log'**)

nwf, 2017-09-11:
`tls-crypt`, the `tls-auth`-on-steroids option, also appears to be missing from the UI (or I just missed it).

Wedmer, 2017-10-03:
I'll check this soon.

Wedmer, 2017-10-09:
resolv_retry exists for client mode.
mssfix exists for udp proto.
tls-crypt is not supported by OpenWrt init script. If you need this option, then open issue for main OpenWrt project. Anyway it will be added.

There're no options for log file except verbosity as all OpenVPN instances use syslog.

PR #1387 (2017-10-10): [luci-app-openvpn] Added support for tls-crypt option.
Should close  #1340
commit bd7eff9a (2017-10-10): [luci-app-openvpn] Added support for tls-crypt option.
Closes #1340
closed 2017-10-10 by hnyman