openwrt/luci #1367: luci-app-ddns - is not possible using when configuration from LuCI interface "wwan0" (mobile modem - ncm connection) - afraid.org* provider

build000, 2017-09-15:
@jow- , @chris5560 , or others
Not possible using in /etc/config/ddns :
```
(...)
option ip_source 'interface'
option interface 'wan'
option ip_interface 'wwan0'
(...)
```
...when configuration from LuCI ddns service - eq. afraid.org* provider (any provider).
"wwan0" does not exist in "box" (https://192.168.1.1/cgi-bin/luci/admin/services/ddns/detail/*) .

chris5560, 2017-09-15:
What version of ddns-scripts / luci-app-ddns are you talking about ?
Latest version of luci-app-ddns - Tab Advanced Settings; if you change IP4/IP6-Source also in the next row all available options are changing to available options.
i.e. setting to "interface" will show "eth0" "eth1" "br-lan"..., so all available interface that are also listed by ifconfig command from console.
There is no dependency when setting ddns-provider in the Basic Settings Tab

build000, 2017-09-15:
@chris5560 
I have described the situation in the latest ddns-script / luci-app-ddns after I have selected a specific provider for this service, ie afraid.org-keyauth.
I just can not select the interface wwan0, because it is not available, so I can not finish LuCI configuration and pop up error. It only works by default, and then after the restart the ddns LuCI service all shows correctly.
And anyway on the margins - ifconfig shows me the wwan0 mobile interface and its parameters.

closed 2023-10-06 by build000