openwrt/luci #1368: luci-app-nlbwmon - not possible view statistic from interface "wwan0" (mobile 4G usb modem - ncm connection)

build000, 2017-09-15:
/etc/init.d/network:
```
(...)
config interface 'wan'
	option name 'wwan0'
	option proto 'ncm'
	option apn 'vpn'
	option pincode '1234'
	option username 'vpn'
	option password 'vpn'
	option ipv6 '0'
	option device '/dev/ttyUSB0'
	option keepalive '10 30'
	option demand '0'
	option metric '0'
	option delegate '0'
	option mode 'auto'
(...)
```

jow-, 2017-09-15:
Provide the output of `ifstatus wan`.

build000, 2017-09-15:
```
root@WITI:~# ifstatus wan
{
	"up": true,
	"pending": false,
	"available": true,
	"autostart": true,
	"dynamic": false,
	"uptime": 21113,
	"l3_device": "wwan0",
	"proto": "ncm",
	"updated": [
		"data"
	],
	"metric": 0,
	"dns_metric": 0,
	"delegation": false,
	"ipv4-address": [
		
	],
	"ipv6-address": [
		
	],
	"ipv6-prefix": [
		
	],
	"ipv6-prefix-assignment": [
		
	],
	"route": [
		
	],
	"dns-server": [
		
	],
	"dns-search": [
		
	],
	"inactive": {
		"ipv4-address": [
			
		],
		"ipv6-address": [
			
		],
		"route": [
			
		],
		"dns-server": [
			
		],
		"dns-search": [
			
		]
	},
	"data": {
		"manufacturer": "huawei"
	}
}
```

jow-, 2017-09-15:
There is no ip address assigned to this interface, therfore nlbwmon cannot select related conntrack traffic.

This needs to be fixed in netifd or the ncm proto.

closed 2017-09-15 by jow-