openwrt/luci #1373: bad login(?) or some new bug in LuCI (theme openwrt)

build000, 2017-09-18:
Reboot (SNAPSHOT, r4866-a23e002)
```
/usr/lib/lua/luci/dispatcher.lua:578: bad argument #1 to 'pairs' (table expected, got nil)
stack traceback:
	[C]: in function 'pairs'
	/usr/lib/lua/luci/dispatcher.lua:578: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:247: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
```

build000, 2017-09-18:
Probably the error was related to the general problem with the 4G mobile connection - I close this issue.
Again more restarts after the upgrade and - working fine.

closed 2017-09-18 by build000

hnyman, 2017-09-18:
I have seen something similar in connection with LuCI menu items that are supposed to be created according to specific app config. E.g. LuCI statistics configures its menu items based on installed plugins, and if the whole statistics config is missing, this kind of error surfaces and you get locked out of LuCI.

I suspect that something similar has been your reason, too.