openwrt/luci #1381: LuCI: invalid login after upgrade

aasgit, 2017-09-30:
I’ve upgraded my router firmware to the latest LEDE snapshot and now I can’t login into LuCI anymore. However, I can ssh just fine (pheew!) and I did not change the root password. I tried with Firefox, Safari and Chrome, to make sure it wasn’t a browser issue. On both MacOS and Linux. At some point, I tried changing the password to ‘password’ just to make sure it wasn’t a special character issue either. No luck. Here are some system details:

root@lede-wyn:~# cat /etc/os-release
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4907-31e9445"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH=“arm_cortex-a9_vfpv3"
LEDE_TAINTS=”"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE=“LEDE Reboot SNAPSHOT r4907-31e9445”

root@lede-wyn:~# opkg list-installed | grep luci
luci - git-17.267.26012-a2ea9fd-1
luci-app-adblock - git-17.267.26012-a2ea9fd-1
luci-app-clamav - git-17.267.26012-a2ea9fd-1
luci-app-ddns - 2.4.8-2
luci-app-firewall - git-17.267.26012-a2ea9fd-1
luci-app-fwknopd - git-17.267.26012-a2ea9fd-1
luci-app-openvpn - git-17.267.26012-a2ea9fd-1
luci-app-sqm - 1.1.3-1
luci-app-uhttpd - git-17.267.26012-a2ea9fd-1
luci-app-vpnbypass - git-17.267.26012-a2ea9fd-1
luci-app-watchcat - git-17.267.26012-a2ea9fd-1
luci-base - git-17.267.26012-a2ea9fd-1
luci-lib-ip - git-17.267.26012-a2ea9fd-1
luci-lib-jsonc - git-17.267.26012-a2ea9fd-1
luci-lib-nixio - git-17.267.26012-a2ea9fd-1
luci-mod-admin-full - git-17.267.26012-a2ea9fd-1
luci-proto-ipv6 - git-17.267.26012-a2ea9fd-1
luci-proto-ppp - git-17.267.26012-a2ea9fd-1
luci-ssl-openssl - git-17.267.26012-a2ea9fd-1
luci-theme-bootstrap - git-17.267.26012-a2ea9fd-1

I'm available for tests and more details.

Please and thank you

aasgit, 2017-10-03:
I have reverted to the last working build and now I can login. This could be a "red light".

current (reverted) LEDE snapshot details:

cat /etc/os-release
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4831-9c500db"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r4831-9c500db"

I hope this helps.

jow-, 2017-10-03:
Unfortunately the commit range is far too huge to draw any conclusions from that. A quick test using a current x86/64 snapshot did not show any login problems either.

Once you're able to reproduce the problem, please check the following things:
 - Is `rpcd` running? `ps ww | grep rpcd`
 - Is Lua available to properly talk to ubus? `lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'`
 - Are logins via ubus working? `ubus call session login '{ "username": "root", "password": "secret" }'`

roburb, 2017-10-09:
Hi I have the same issue. I tried to find solution alone, but without success.
I have run the commands with following results:

`ps ww | grep rpcd`
`802 root      1440 S    /sbin/rpcd`
`18991 root      1184 S    grep rpcd`

`lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'`
`board_name      tl-wdr4300`
`hostname        d31785`
`release table: 0xb549f0`
`        distribution    LEDE`
`        revision        r3533-d0bf257c46`
`        version 17.01.3`
`        codename        reboot`
`        target  ar71xx/generic`
`        description     LEDE Reboot 17.01.3 r3533-d0bf257c46`
`model   TP-Link TL-WDR4300 v1`
`kernel  4.4.89`
`system  Atheros AR9344 rev 2`
`ubus call session login '{ "username": "root", "password": "……." }'`
`{`
 `       "ubus_rpc_session": "525f7827cfdb95eaf9726d5db67e7677",`
 `       "timeout": 300,`
 `       "expires": 299,`
 `       "acls": {`
 `               "access-group": {`
 `                       "unauthenticated": [`
 `                               "read"`
 `                       ]`
 `               },`
 `               "ubus": {`
 `                       "session": [`
 `                               "access",`
 `                               "login"`
 `                       ]`
 `               }`
 `       },`
 `      "data": {`
 `               "username": "root"`
 `       }`
`}`

What can I do now, to find the solution?
Thank You

mdbuerkle, 2017-10-31:
You might want to check out issue#1101 - maybe there's (a hint to) Your solution?

aasgit, 2017-11-11:
@jow- 

I thought I gave the latest build a try and this is still happening. here's the details (including your commands output):

```
root@lede-wyn:~# date
Sat Nov 11 15:07:33 CET 2017

root@lede-wyn:~# cat /etc/os-release
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r5320-b4e30b863c"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r5320-b4e30b863c"

root@lede-wyn:~# ps ww | grep rpcd
 4730 pts/0    S+     0:00 grep rpcd

root@lede-wyn:~# lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'
board_name	armada-385-linksys-rango
hostname	lede-wyn
release	table: 0xa0b8c0
	distribution	LEDE
	revision	r5320-b4e30b863c
	version	SNAPSHOT
	codename	reboot
	target	mvebu/generic
	description	LEDE Reboot SNAPSHOT r5320-b4e30b863c
model	Linksys WRT3200ACM
kernel	4.9.58
system	ARMv7 Processor rev 1 (v7l)

root@lede-wyn:~# ubus call session login '{ "username": "root", "password": "my secret password here" }'
{
	"ubus_rpc_session": "faf4bcf4a6eeb220da17f48bfa555a18",
	"timeout": 300,
	"expires": 300,
	"acls": {
		"access-group": {
			"unauthenticated": [
				"read"
			]
		},
		"ubus": {
			"session": [
				"access",
				"login"
			]
		}
	},
	"data": {
		"username": "root"
	}
}
```

I hope these details can help troubleshooting.

I will keep the current latest custom built image installed to provide more details for a few days. Please ask away before I rollback to a working custom built image.

Cheers

gacallea, 2017-12-01:
@jow- 

FWIW: installed latest build just now, still the same issue. here's some details/commands output:

```
root@lede-wyn:~# date
Fri Dec  1 11:16:13 CET 2017

root@lede-wyn:~# cat /etc/os-release
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r5442-b0b289e"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r5442-b0b289e"

root@lede-wyn:~# ps ww | grep rpcd
 5180 pts/0    S+     0:00 grep rpcd

root@lede-wyn:~# lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'
board_name	armada-385-linksys-rango
hostname	lede-wyn
release	table: 0x130a8c0
	distribution	LEDE
	revision	r5442-b0b289e
	version	SNAPSHOT
	codename	reboot
	target	mvebu/generic
	description	LEDE Reboot SNAPSHOT r5442-b0b289e
model	Linksys WRT3200ACM
kernel	4.9.65
system	ARMv7 Processor rev 1 (v7l)

root@lede-wyn:~# ubus call session login '{ "username": "root", "password": "my secret passwod" }'
{
	"ubus_rpc_session": "61d4fccd0e6169340918f2f3afdf6487",
	"timeout": 300,
	"expires": 300,
	"acls": {
		"access-group": {
			"unauthenticated": [
				"read"
			]
		},
		"ubus": {
			"session": [
				"access",
				"login"
			]
		}
	},
	"data": {
		"username": "root"
	}
}
```

please ask away for anything useful for me to help you fix it. this issue has been going on since September.

gacallea, 2017-12-30:
me, again.

```
root@lede-wyn:~# date
Sat Dec 30 21:39:17 CET 2017

root@lede-wyn:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r5645-7aa15953e1"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS=""
LEDE_DEVICE_MANUFACTURER="OpenWrt"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="OpenWrt SNAPSHOT r5645-7aa15953e1"

root@lede-wyn:~# ps ww | grep rpcd
 5202 pts/0    S+     0:00 grep rpcd

root@lede-wyn:~# lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))
> ^C
root@lede-wyn:~# lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'
board_name	armada-385-linksys-rango
hostname	lede-wyn
release	table: 0x13918c0
	distribution	OpenWrt
	revision	r5645-7aa15953e1
	version	SNAPSHOT
	target	mvebu/generic
	description	OpenWrt SNAPSHOT r5645-7aa15953e1
model	Linksys WRT3200ACM
kernel	4.9.72
system	ARMv7 Processor rev 1 (v7l)

root@lede-wyn:~# ubus call session login '{ "username": "root", "password": "my secret password" }'
{
	"ubus_rpc_session": "9b4b339c79666757288ca88084016bec",
	"timeout": 300,
	"expires": 300,
	"acls": {
		"access-group": {
			"unauthenticated": [
				"read"
			]
		},
		"ubus": {
			"session": [
				"access",
				"login"
			]
		}
	},
	"data": {
		"username": "root"
	}
}
```

[image]

limpkin, 2018-04-29:
getting the same issue with current git master

limpkin, 2018-04-29:
Getting the very same error... but only after the first device reboot.

limpkin, 2018-04-29:
ok... found the solution. In my case this was due to a corrupted /etc/config/rpcd. Deleted it and added the default contents:

config login
        option username 'root'
        option password '$p$root'
        list read '*'
        list write '*'

citakis, 2018-07-22:
I had the same issue on raspberry pi model B with lede-17.01.4, lede-17.01.5 and openwrt-18.06.0-rc2.
The solution with the rpcd file did not worked for me because it had the default content.
I can connect with ssh and at Luci interface when I enter wrong credentials it writes "Invalid username and/or password! Please try again." but with the correct credentials it does nothing, no message no login just stays at the login page.
The `ps ww | grep rpcd`  output is
>193 root      1244 S    /sbin/rpcd
  553 root      1052 S    grep rpcd

`lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))'` output is
>board_name	rpi-b
hostname	LEDE
release	table: 0xc6c4b0
	distribution	LEDE
	revision	r3560-79f57e422d
	version	17.01.4
	codename	reboot
	target	brcm2708/bcm2708
	description	LEDE Reboot 17.01.4 r3560-79f57e422d
model	Raspberry Pi Model B Rev 1
kernel	4.4.92
system	ARMv6-compatible processor rev 7 (v6l)

and `ubus call session login '{ "username": "root", "password": "secret" }'` output
>{
	"ubus_rpc_session": "7002909ca9433066689a99993b35072f",
	"timeout": 300,
	"expires": 300,
	"acls": {
		"access-group": {
			"unauthenticated": [
				"read"
			]
		},
		"ubus": {
			"session": [
				"access",
				"login"
			]
		}
	},
	"data": {
		"username": "root"
	}
}

Unfortunately so far I was not able to troubleshoot the issue by myself, so any ideas are more than welcome.

citakis, 2018-07-22:
I just figured that the problem appears when I change lan ip from 192.168.1.1 to another subnet ip for example 192.168.7.1

citakis, 2018-07-31:
Problem solved for me. I cleared browser's cache and the luci made the login and loaded the interface without a problem. It took me two weeks trying every possible combination in order to solve this and finally it was the simplest of all - clear the cache!!!

NightA, 2018-09-10:
I get the same problem with the latest version of Luci in OpenWRT 18.06.1's stable repository (git-18.247.71242) when using and an additional password prompt through uhttpd. 
Disabling the said prompt fixes it, however now i can't use the feature without having it break Luci's session again.

cshoredaniel, 2018-09-11:
Have you tried clearing the cache, but making sure that it removes 'Active Logins' not just cookies (some browsers don't do that by default).

NightA, 2018-09-11:
Yes i did. Also tried using incognito and separate browsers with a clean instances.

gretel, 2019-05-16:
i have to use a numerical ip address to be able to login.

virivongithub, 2019-11-06:
hangs on login with any credentials
ps ww | grep rpcd output: 
718 root      1528 S    /sbin/rpcd -s /var/run/ubus.sock -t 30
 1715 root      1200 S    grep rpcd

lua -lluci.util -lubus -e 'luci.util.dumptable(ubus.connect():call("system", "board", {}))' output:
board_name	tl-wr1043nd-v2
hostname	OpenWrt
release	table: 0xa421a0
	distribution	OpenWrt
	revision	r7808-ef686b7292
	version	18.06.4
	target	ar71xx/generic
	description	OpenWrt 18.06.4 r7808-ef686b7292
model	TP-Link TL-WR1043N/ND v3
kernel	4.9.184
system	Qualcomm Atheros QCA9558 ver 1 rev 0

ubus call session login '{ "username": "root", "password": "secret" }' output:
{
	"ubus_rpc_session": "fc2f8b66be4723bab2f7dd77e3c85b00",
	"timeout": 300,
	"expires": 300,
	"acls": {
		
	},
	"data": {
		"username": "root"
	}
}

virivongithub, 2019-11-06:
got in but now its giving me this error https://i.imgur.com/AZi4vjq.png (cleared cache)

jow-, 2020-02-23:
Not able to reproduce on master or 19.07 anymore

closed 2020-02-23 by jow-