openwrt/luci #1388: installing luci on ubuntu

fanxiangqiang, 2017-10-10:
i am so sorry for a question.i want to install luci on ubunu,and i download luci source code into my folder, next i  cd luci ,tap "make runuhttpd",then the trouble shows! im so sorry! because  of svn service stopping,but some code is needed to be downloaded from svn,i am upset. im thankful who knowns the solution!

make[1]: Leaving directory `/home/fanxiangqiang/luci/luci-luci-0.12/contrib/uci'
make[1]: Entering directory `/home/fanxiangqiang/luci/luci-luci-0.12/contrib/uhttpd'
svn co svn://svn.openwrt.org/openwrt/branches/backfire/package/uhttpd/src uhttpd-src
svn: E000111: Unable to connect to a repository at URL 'svn://svn.openwrt.org/openwrt/branches/backfire/package/uhttpd/src'
svn: E000111: Can't connect to host 'svn.openwrt.org': Connection refused
make[1]: *** [uhttpd-src/Makefile] Error 1
make[1]: Leaving directory `/home/fanxiangqiang/luci/luci-luci-0.12/contrib/uhttpd'
*** Compilation of contrib/uhttpd failed!

fanxiangqiang, 2017-10-10:
[image]

hnyman, 2017-10-10:
You are looking at old ancient unsupported version of Luci (luci-0.12). there is no more svn server etc.

I am not sure if that kind of separate installation is supported, any more.

fanxiangqiang, 2017-10-10:
Dose it mean  i cannot install    luci  on my ubuntu! ......,i will try it once more

Wedmer, 2017-10-10:
Luci needs OpenWrt uci. There're no bindings for uci in ubuntu.

kdm6389, 2017-10-28:
You can work with LEDE + LuCI on VirtualBox, I guess.

hnyman, 2017-10-28:
I close this, as basically LuCI is not meant to be run outside the Openwrt/LEDE environment. It is tightly tied to that.

closed 2017-10-28 by hnyman

syedumerbukhari, 2019-07-26:
> i am so sorry for a question.i want to install luci on ubunu for instantly programming and debugging ,and i download luci source code into my folder, next i cd luci ,tap "make runuhttpd",then the trouble shows! im so sorry! because of svn service stopping,but some code is needed to be downloaded from svn,i am upset. im thankful who knowns the solution!
> 
> make[1]: Leaving directory `/home/fanxiangqiang/luci/luci-luci-0.12/contrib/uci' make[1]: Entering directory `/home/fanxiangqiang/luci/luci-luci-0.12/contrib/uhttpd'
> svn co svn://svn.openwrt.org/openwrt/branches/backfire/package/uhttpd/src uhttpd-src
> svn: E000111: Unable to connect to a repository at URL 'svn://svn.openwrt.org/openwrt/branches/backfire/package/uhttpd/src'
> svn: E000111: Can't connect to host 'svn.openwrt.org': Connection refused
> make[1]: *** [uhttpd-src/Makefile] Error 1
> make[1]: Leaving directory `/home/fanxiangqiang/luci/luci-luci-0.12/contrib/uhttpd'
> *** Compilation of contrib/uhttpd failed!

brother did you found the solution on this issue?