openwrt/luci #1395: can't set protocol wireguard on the interface / lede-17.01.3

gorsha, 2017-10-17:
Last lede says Wireguard "Protocol not supported":

Tue Oct 17 08:04:28 2017 daemon.notice netifd: RAG (24892): Unable to set device: Protocol not supported
Tue Oct 17 08:04:33 2017 daemon.notice netifd: Interface 'RAG' is now down
Tue Oct 17 08:04:33 2017 daemon.notice netifd: Interface 'RAG' is setting up now
<.. and so on infinite times..>

Linux version 4.4.89 (buildbot@builds-02.infra.lede-project.org) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3101-bce140e) ) #0 Sat Sep 30 13:27:29 2017
CPU0 revision is: 0001974c (MIPS 74Kc)
SoC: Atheros AR9344 rev 2
Hardware : D-LINK WDR3600

root@lede:~# opkg install kmod-wireguard wireguard-tools
Package kmod-wireguard (4.4.89+0.0.20170115-1) installed in root is up to date.
Package wireguard-tools (0.0.20171011-1) installed in root is up to date.

hnyman, 2017-10-17:
You will likely need to wait for the forthcoming 17.01.4 release, or build 17.01 by yourself.
Wireguard has been recently upgraded and moved to the core packages, so 17.01.3 users are a bit out of sync (as user-space package is newer than the underlying kernel package)

cc @zx2c4

zx2c4, 2017-10-17:
@hnyman is right: wait for 17.01.4, which will solve all this.

> Package kmod-wireguard (4.4.89+0.0.20170115-1) installed in root is up to date.
> Package wireguard-tools (0.0.20171011-1) installed in root is up to date.

Moving forward, kmod-wireguard >= 20171005 will work with all wireguard-tools >= 20171005, so we shouldn't encounter this in the future.

This ticket can be closed.

closed 2018-03-07 by aparcar

psy21d, 2018-04-04:
how about this problem now?
how i check version, how to build newest version wireguard on lede?