openwrt/luci #1410: Apostrophe (UTF-8: &#39;) shows as code instead of character in luci using french translation

blamaz, 2017-10-22:
(Overview) translated to: (Vue d&#39;ensemble) 
(Hosts names) translated to (Noms d&#39;hôtes)
Tried the following with same results:
Chrome and IE on 2 differents computers,
Lede 17.0.4 stable branch and Davidc502 fork,
Forced Chrome to use UTF-8 instead of automatic detection.

Thank for your work

kyle30312, 2018-09-27:
This problem occurs with Bootstrap and Material themes.  It does not occur with Freifunk_Generic and OpenWrt themes.

[image]

[image]

[image]

[image]

closed 2018-09-27 by jow-

viking76, 2019-03-13:
We have the same problem with 18.06.2, why it is Closed issue. Thanks for french users :)

kyle30312, 2019-03-14:
@jow- Perhaps fixes to bootstrap and material themes need to be merged from master to openwrt-18.06 branch?

viking76, 2019-03-16:
@jow excuse for my wrong information (last snapshot tested), as kyle i think it can be pushed to openwrt-18.06,X, i have tested all theme all theme just rose-theme have  problem. thanks this snapshot really stable (more than 6 month).

commit 72d4b62d (2018-09-27): luci-theme-material: do not double escape menu titles

Fixes #1410.
commit ce46437b (2018-09-27): luci-theme-bootstrap: do not double escape menu titles

Fixes #1410.