openwrt/luci #1412: luci-app-statistics: apcups plugin doesn't show "Voltages on APC UPS - AC" graph when /tmp/rrd/TP-Link/apcups/voltage-output.rrd is not generated.

stephengeorgewest, 2017-10-24:
I have a "MODEL: Back-UPS ES 550G" ups, and it doesn't report output voltage, and thus voltage-output.rrd` is not generated. This seems to prevent the whole graph from showing up. It started showing up after I copied voltage-input.rrd to voltage-output.rrd. 

    root@TP-Link:~# cp /tmp/rrd/TP-Link/apcups/voltage-input.rrd /tmp/rrd/TP-Link/apcups/voltage-output.rrd

Is this the expected behavior for luci-app-statistics? or is the `collectd-mod-apcups | 5.5.3-4` opkg expected to fill that with empty values?

The long tale.

Over the weekend I upgraded to LEDE from Chaos Calmer. I thought this would be a good time to get apcupsd working. I first found: https://wiki.openwrt.org/doc/howto/apcupsd_es500 . It had some typos and missing dependencies, but I powered through most of it. I got apcaccess to work, and my pc could connect to the apcupsd server. But I wanted the pretty graphs. I searched some more and found: https://nordicboycomputing.com/index.php?post/2015/04/06/Patch-to-enable-graphs-on-luci-app-statistics-for-collectd-mod-apcups . I applied those patches. Restarted. Graphs! I thought maybe I'll be an internet hero and fix it. I look up the source and find:  808f2617931dcd6f84637e1bff3a737b9368917c . Already there ... oh, in master, not built yet. Ok, I'll just grab the latest version and apply those changes. 2145b7b33bbc4c142b243ac032e6e9c1c28ca835 , The battery one showed up. That wasn't there before. Still missing "Voltages on APC UPS - AC". google ... ls ... vi ... apcaccess ... google ... http://www.apcupsd.org/manual/manual.html#apcupsd-status-logging ... ls ... google ... https://wiki.openwrt.org/doc/howto/statistics.apcupsd_multimeter ... bobmseagithub writes data to `voltage-output.rrd` ... :suspect: ... ls ... rrdtool dump ... ls ... :thought_balloon: ... cp ...  Now here I am typing all of this up. And now I realize that bobmseagithub is probably @bobmseagithub

bobmseagithub, 2017-12-02:
Did you get it working?  FYI: These links might be helpful:
https://github.com/openwrt/luci/pull/1227   ( I think you found that one already )
https://github.com/openwrt/luci/pull/1240
https://wiki.openwrt.org/doc/howto/apcupsd_su700  ( a new page I just wrote for an older UPS ).

Hope that helps!
-Bob

guidosarducci, 2018-03-02:
@stephengeorgewest  Based on your experiences you will probably be interested in #1653, which now handles some data not being reported by various UPSes.

Cheers,
Tony

hnyman, 2018-03-09:
Closing as the issue is likely fixed by #1653

closed 2018-03-09 by hnyman

stephengeorgewest, 2018-06-24:
That looks like it fixes my issue, thanks.