openwrt/luci #1419: VLAN can be deleted even though working in an interface

HenryGiraldo, 2017-10-30:
It's not really a bug, but it would be helpful to use poka-yoke here.

It can leave you without access to the device and you are forced to reset it from the factory or worse still having to use the serial port to recover the device.

jow-, 2017-10-30:
Yeah but in general there's many way to lock yourself out. You can also disable dropbear, delete lan, break the firewall rules, ...

HenryGiraldo, 2017-10-30:
Yes, that's true, but I think it's a good practice to check if something is in use before you can delete it. 

In the case of dropbear you already know that as direct implication you will lose access through ssh, in this case you are destroying an interface configuration as collateral damage.

metronidazole, 2018-11-23:
Is this still an issue with that feature that automatically reverts configuration changes that break your browser's access to LuCi?

HenryGiraldo, 2018-11-26:
You could have access through the lan interface and not through the selected vlan. I don't think it's so much a problem related to the access itself, but to the robust of the system. It's a potential improvement, it's not really a "failure".

closed 2019-08-22 by HenryGiraldo