openwrt/luci #1420: When an interface is deleted, it continues appearing in network tab

HenryGiraldo, 2017-10-30:
Initial state:
[image]
Creating the interface:
[image]
Interface working:
[image]
After pressing the delete button:
[image]

woutput, 2018-06-14:
I ran into the same issue.
Note that in my case, deleting an interface
- works correctly when using Firefox on laptop
- does not work correctly when using Chrome on Android smartphone

So this may be browser related(?)

closed 2019-08-22 by HenryGiraldo

woutput, 2019-08-22:
Solved? If so, in which commit/release?
Thanks

HenryGiraldo, 2019-08-22:
I don't know. It's been too long, I don't find any reason to keep the ticket open.