openwrt/luci #1424: Request for integer Y-axis dividers in Realtime Graphs

richb-hanover, 2017-11-01:
In the Realtime Graphs section, the (quite attractive) graphs have peculiar Y-axis values. For example, in the image below, the top value is 831.6kbps, and the sub-dividers are at 554.4kbps and 277.2kbps.

[image]

I propose that the max value be rounded to integer values (times a power of 10). In the image above, the dividers could be at 900, 600 & 300 kbps with no loss of accuracy. It's always hard to interpolate the values from the graph. But it would be easier to think about/form a mental image of “almost 900 kbps” than “almost 831.6kbps”.

To flesh out the proposal:

* If the max value is close to 1.0 (or a factor of 10 times 1.0), there could be four dividers labeled at 0.25, 0.5, 0.75, and 1.0.

* 2.0 - four dividers at 0.5, 1.0, 1.5, and 2.0

* 3.0 - three dividers at 1.0, 2.0, and 3.0

* 4.0 - four dividers at 1.0, 2.0, 3.0, and 4.0

* 5.0 - five dividers at 1.0, 2.0, 3.0, 4.0, and 5.0

* 6.0 - three dividers at 2.0, 4.0, and 6.0

* (I would skip 7, and use 6.0 or 8.0 scales, whichever is closer)

* 8.0 - four dividers at 2.0, 4.0, 6.0 and 8.0

* 9.0 - three dividers at 3.0, 6.0, and 9.0

hnyman, 2017-11-01:
Rich,
feel free to propose a patch to implement that...

At the first glance it looks like the found peak should be rounded up:
https://github.com/openwrt/luci/blob/44bf3f0c1640040561306caff39f34ac916b4357/modules/luci-mod-admin-full/luasrc/view/admin_status/bandwidth.htm#L196

hnyman, 2017-11-01:
Some googling revealed that to rounding N to e.g. X significant digits can be done in excel with
ROUND(number,digits-(1+INT(LOG10(ABS(number)))))

so converting that to mode reduced javascript logic is the task...

E.g. 110% of the max value rounded to 2 significant digits could offer a nice starting base

commit 7b43e672 (2018-07-03): luci-mod-admin-full: improve Y-axis scaling in bandwidth graph

Calculate a peak that causes the vertical scale to use even values.

Fixes #1424.
closed 2018-07-03 by jow-