openwrt/luci #1431: static routes lost after reboot

nixmomo, 2017-11-09:
hi,

i have set some static route via luci.
the target is to route some ip's to a vpn tunnel on a new interface... all works as expected but after a reboot the routes are all lost.
in luci webinterface are the routes still persent but i cant activate it again. if i click on apply again the routes won't be set again. I have to delete the routes and set them again to bring it up to work.

the sames as described here
https://forum.lede-project.org/t/static-route-not-working/5993/8
here is my config
`config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd7d:3b4e:f31c::/48'

config interface 'lan'
	option type 'bridge'
	option ifname 'eth0.1'
	option proto 'static'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option ipaddr '192.168.1.1'

config interface 'wan'
	option ifname 'eth0.2'
	option proto 'dhcp'
	option peerdns '0'
	option hostname 'router'

config interface 'wan6'
	option ifname 'eth0.2'
	option proto 'dhcpv6'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '0t 2 3 4 5'
	option vid '1'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '0t 1'
	option vid '2'

config interface 'dlhost'
	option proto 'none'
	option ifname 'tun0'

config route
	option interface 'dlhost'
	option target '148.251.19.108'

config route
	option interface 'dlhost'
	option target '148.251.220.129'

config route
	option interface 'dlhost'
	option target '148.251.154.115'

config route
	option interface 'dlhost'
	option target '138.201.20.15'

config route
	option interface 'dlhost'
	option target '138.201.169.4'`

how can i fix this?

feckert, 2018-04-06:
I think on boot the VPN-Interface is not set up and so is not connected to the other Network.
Because of this the static routes in the system are not written to the routing table.
If you do a /etc/init.d/network restart the routings are setup.

nixmomo, 2018-04-06:
Nope.. other internal static routes doesn't work too

jow-, 2018-07-03:
CLosing as this is unrelated to LuCI.

closed 2018-07-03 by jow-