openwrt/luci #1439: Unneeded secret in openvpn config

windowsrefund, 2017-11-13:
opkg reports I have the following package installed: luci-app-openvpn - git-17.315.61305-a9bddc4-1

I have configured my openvpn instance and the service fails to start. This error shows up in the system log:

`Mon Nov 13 22:32:23 2017 daemon.err openvpn(pia_client)[13823]: Options error: specify only one of --tls-server, --tls-client, or --secret`

Upon closer inspection, I notice the following line in /var/etc/openvn-pia_client.conf

`secret shard-secret.key`

This line appears to be written to the file despite the fact I never intentionally add it. Removing the line and attempting to start the openvpn service works but the line is written to the file every time I attempt to start the instance from the LUCI interface.

windowsrefund, 2017-11-13:
Unable to duplicate.

windowsrefund, 2017-11-13:
Thought I couldn't reproduce but it showed up again after reconfiguring from scratch.

BradleyDS2, 2017-11-16:
Happening to me as well, it adds it every time the router boots.

rik-shaw, 2018-01-17:
I have automated the removal of the 'secret' line with the following file:

/etc/init.d/openvpn_secret_patch
```
#!/bin/sh /etc/rc.common
# (C) 2018 Rik Shaw

START=99

boot() {
	sed -i '/secret/d' /var/etc/openvpn-pia_client.conf
}
start() {
	sed -i '/secret/d' /var/etc/openvpn-pia_client.conf  
}
```
Then enable and start the service:
```
/etc/init.d/openvpn_secret_patch enable
/etc/init.d/openvpn_secret_patch start
```

jow-, 2018-01-17:
Remove the `option secret` from /etc/config/openvpn

rik-shaw, 2018-01-17:
@jow- now that it a proper fix: thanks!

BradleyDS2, 2018-01-18:
To anyone else like me who can't CD into /etc/config I found that you can if you do the following:
cd /etc
nano config

I'll say that config is a directory, then close nano.
cd config

You should now be in /etc/config

No idea why I can't access it directly, but this workaround seems to work. Not sure if other text editors work, I've only tried nano.