openwrt/luci #1442: Using OpenWrt 15.05.1 can't login when LAN set to 192.168.0.x

AmazingAdrian, 2017-11-15:
So, plug in laptop to LAN port, gets a 192.168.1.x address. Webpage to 192.168.1.1. Show Luci text then redirects to login screen. Click Login and get in. Change the LAN static IP address from 192.168.1.1 to 192.168.0.1 (to sit with my existing LAN), click Save and Apply. Lose the webpage (to be expected), from my laptop renew IP, get a 192.168.0.x address, open webpage to 192.168.0.1 and it just sits on the LUCI text screen. Click on that and get the login screen. Click Login and it rejects it. I've also tried by setting a password before changing the IP address. Same result.
If I set the IP address to 192.168.2.1 or .3.1 etc it's fine. Is this a 'feature' that I should know about?

AmazingAdrian, 2017-11-16:
Well, new day, new issues !!! It now logs in (but slowly) but after changing any setting (normally WiFi based), I get logged out :-0

jow-, 2017-11-16:
Sounds like you have an IP address conflict or something in your LAN. There is no special handling of `192.168.0.1` anywhere in OpenWrt, LEDE or LuCI.

closed 2017-11-16 by jow-