openwrt/luci #1489: Typos

bugmenot458, 2017-12-28:
https://github.com/openwrt/luci/blob/master/documentation/api/modules/nixio.fs.html#L1277
> It is normally not possible to rename files across **fileystems**.

<br/>

https://github.com/openwrt/luci/blob/master/libs/luci-lib-nixio/docsrc/nixio.fs.lua#L50
> It is normally not possible to rename files **accross fileystems**.

<br/>

https://github.com/openwrt/luci/blob/master/modules/luci-mod-admin-full/luasrc/model/cbi/admin_system/fstab.lua#L65
> o = s:option(Flag, "check_fs", translate("Check **fileystems** before mount"), translate("Automatically check filesystem for errors before mounting"))

hnyman, 2017-12-28:
fixed in master with 878a64f5386d5eabf2bf81bdc061f8d7d238c7f8

closed 2017-12-28 by hnyman

dartraiden, 2018-01-02:
@hnyman, "accross" should be fixed as well