openwrt/luci #1507: DNS servers on Status Overview page

patricktokeeffe, 2018-01-07:
> Powered by LuCI (git-15.248.30277-3836b45) / OpenWrt Chaos Calmer 15.0

I changed my DNS servers to use Google addresses (8.8.8.8, 8.8.4.4) but the server addresses displayed on the Status > Overview page still reflect my ISPs servers:

[image]

I'm pretty certain that Google's servers are actually being used: I get errors for non-existent domains instead of an ISP search page, and the ISP addresses don't show up in `dig` traces. 

Since this is a "status" page, I expect the actual, not advertised, DNS servers be listed here

jow-, 2018-01-07:
HOw did you change them?

patricktokeeffe, 2018-01-07:
I set them under *Network* > *DHCP and DNS*. Is there another place they can be configured?

jow-, 2018-01-07:
Yes. Admin -> Network -> Interface -> wan -> Advanced Settings.
Uncheck `Use DNS servers advertised by peer` and fill in one or more addresses into `Use custom DNS servers`

patricktokeeffe, 2018-01-07:
> Uncheck `Use DNS servers advertised by peer`

priceless 💪 thank you @jow-

closed 2018-01-07 by patricktokeeffe