openwrt/luci #1511: Luci configurable parameters for Mesh (11s)

bobster-galore, 2018-01-08:
Would be nice to configure parameters of Mesh (11s) within Luci, minimum "option mesh_fwding '0'".

bobster-galore, 2018-01-08:
s.a. https://github.com/openwrt/luci/pull/1513 concerning mesh forwarding.

kishangondaliya, 2018-01-09:
@bobster-galore ,
Can you list down working params with possible values.
By working I mean, when you use it in /etc/config/wireless, mesh interface should be configured as expected.
Also list down params which are not supported and needs to be integrated.

SvenRoederer, 2018-01-09:
"option mesh_fwding" was merged in https://github.com/openwrt/luci/pull/1513

bobster-galore, 2018-01-09:
Today I changed the mesh-id several times until I found a satisfying string for berlin community (mesh-ch13.berlin.freifunk.net) and out of sudden one of the routers didn't mesh any more.
When I checked /etc/config/wireless I found "option key '1'" - any ideas where does that come from?
May be "option mesh_fwding" is enough, we should discuss that.

kishangondaliya, 2018-01-10:
Available mesh params with their default values.

mesh_retry_timeout = 100 milliseconds 
mesh_confirm_timeout = 100 milliseconds 
mesh_holding_timeout = 100 milliseconds 
mesh_max_peer_links = 32 
mesh_max_retries = 3 
mesh_ttl = 31 
mesh_element_ttl = 31 
mesh_auto_open_plinks = 1 
mesh_hwmp_max_preq_retries = 4 
mesh_path_refresh_time = 1000 milliseconds 
mesh_min_discovery_timeout = 100 milliseconds 
mesh_hwmp_active_path_timeout = 5000 TUs 
mesh_hwmp_preq_min_interval = 10 TUs 
mesh_hwmp_net_diameter_traversal_time = 50 TUs 
mesh_hwmp_rootmode = 0 
mesh_hwmp_rann_interval = 5000 TUs 
mesh_gate_announcements = 0 
mesh_fwding = 1 
mesh_sync_offset_max_neighor = 50 
mesh_rssi_threshold = 0 dBm 
mesh_hwmp_active_path_to_root_timeout = 6000 TUs 
mesh_hwmp_root_interval = 5000 TUs 
mesh_hwmp_confirmation_interval = 2000 TUs 
mesh_power_mode = active 
mesh_awake_window = 10 TUs 
mesh_plink_timeout = 1800 seconds 

As a part of 802.11s community, I would like all of them available on OpenWRT.

SvenRoederer, 2018-01-10:
according to https://github.com/openwrt/openwrt/blob/d91494eedf06ac6b31c1aa9f7172871b16af96c8/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh at least most of them have a CUI-setting on OpenWRT.
But if you want to make them all avail in LuCI, this might create a mess on the web-site. So probably an extra page might be the best.

FreifunkUFO, 2018-01-12:
thats a big list, but i think the next most important value for 11s to add is: **mcast_rate**
the ability to change it via luci is also missing for ad-hoc (please add it for that too)

and next step: we could also add values to set an encrypted 11s. but then wpad-mesh (and authsae) is mandatory needed. so hints or queries might to be added (whether these packages are installed or not).

bobster-galore, 2018-01-13:
What about introducing to Luci what @FreifunkUFO mentions and leaving the rest @ config-files?
I believe those who want to alter the less common options know how to use the config-files for their purpose.

@kishangondaliya likes to have all the mesh-options available on openwrt - from https://wiki.openwrt.org/doc/howto/mesh.80211s to https://dev.openwrt.org/browser/trunk/package/kernel/mac80211/files/lib/netifd/wireless/mac80211.sh I think they are all available thru the network driver.

Perhaps it would be nice to have a default config file having all the by @kishangondaliya above mentioned settings - so everybody can feel free to alter the defaults. ... and a hint where to alter the settings.

I can imagine this could be a nice way to deal with Mesh (11s), what do u think?

adelton, 2025-01-10:
Am I correct in assuming that the code now lives in https://github.com/freifunk/openwrt-packages so this issue should be moved there and / or closed in this repo?