openwrt/luci #1512: Luci correct display of existing Mesh (11s)

bobster-galore, 2018-01-08:
At luci/admin/network/wireless survey would be nice to see correct status of Mesh (11s) wifi including mesh_id.

bobster-galore, 2018-06-28:
Has there been any progress on this subject?
Since some devices don't support IBSS-mode any more it would be really nice to have a working configuration and display of mesh (11s).
What could be a support for some progress?
@SvenRoederer

bobster-galore, 2018-06-28:
s.a. https://github.com/freifunk-berlin/firmware/issues/536
https://github.com/freifunk-berlin/firmware/issues/576

jow-, 2018-07-03:
I'd be willing to spend some time on this if you can get me SSH access to an existing 11s setup.

pmelange, 2018-07-05:
@jow- I could set up temporarily a couple of nodes which mesh with each other via 11s.  Could you please give me your public ssh key?  

Please note that we (freifunk) set "internal forwarding of Mesh-peers" to disabled and rely on OLSRd to do the routing.

I will try to set up the nodes tonight.

pmelange, 2018-07-05:
The nodes are set up and ready to be used.  @bobster-galore, do you also want access?  If so, send me you public ssh key.

bobster-galore, 2018-07-05:
@jow- Shall I describe what is not functioning?

Probably u are also close to the display of disabled AP
https://github.com/freifunk-berlin/firmware/issues/501

pmelange, 2018-07-22:
I have taken down the test setup.  There's no use in wasting electricity when nobody shows any interest

jow-, 2018-07-22:
@pmelange - I didn't even notice your offer here. You can find my pubkey at http://mein.io/

pmelange, 2018-07-23:
@jow- Take a look at the file "/root/info.txt"

jow-, 2018-07-23:
@pmelange - thanks, so I begun taking a look around. First problem I see is that there's apparently no nl80211 api for retrieving the effective mesh id, so we have to read it from ubus state info and hope that it is sync.

bobster-galore, 2018-07-23:
@pmelange thankes!
@jow- isn't it there https://git.kernel.org/pub/scm/linux/kernel/git/linville/wireless.git/tree/include/uapi/linux/nl80211.h?id=HEAD#n1079

jow-, 2018-07-23:
@bobster-galore: hm, not sure what exactly you mean with the link... can you elaborate?

bobster-galore, 2018-07-23:
I understood u were looking for the mesh_id and how to retrieve it?
I thought u can ask for that variable @NL80211_ATTR_MESH_ID with @NL80211_CMD_GET_MESH_CONFIG https://git.kernel.org/pub/scm/linux/kernel/git/linville/wireless.git/tree/include/uapi/linux/nl80211.h?id=HEAD#n270
or is that far from where u program?

jow-, 2018-07-23:
Will give it a try tomorrow- so far I only experimented with „iw ... get mesh_param“ and that didn‘t expose any ID info

bobster-galore, 2018-07-23:
Indeed iw can set but not display mesh_id, weird. 
like that ```iw phy phy0 interface add mesh0 type mp mesh_id mymesh```
Here ```uci show wireless.@wifi-iface[3].mesh_id``` gives the mesh_id.

jow-, 2018-07-24:
@pmelange - the test setup seems to be down again

pmelange, 2018-07-24:
@jow- The system is back online.  I had to reboot the router.  Wan wasn't getting an IP addr, and "ifup wan" was hanging.

logread (last line repeated through the entire logbuffer)
```
Tue Jul 24 16:42:48 2018 authpriv.info dropbear[19105]: Child connection from 213.61.250.146:44784
Tue Jul 24 16:42:49 2018 authpriv.notice dropbear[19105]: Pubkey auth succeeded for 'root' with key md5 88:9b:b0:e2:ee:3f:92:68:c3:71:ba:ee:ce:0a:e3:06 from 213.61.250.146:44784
Tue Jul 24 16:43:08 2018 kern.info kernel: [101417.008290] 
Tue Jul 24 16:43:08 2018 kern.info kernel: [101417.008290] do_page_fault(): sending SIGSEGV to netifd for invalid read access from 00000000
Tue Jul 24 16:43:08 2018 kern.info kernel: [101417.017039] epc = 0040c321 in netifd[400000+17000]
Tue Jul 24 16:43:08 2018 kern.info kernel: [101417.022040] ra  = 0040cd09 in netifd[400000+17000]
Tue Jul 24 16:43:08 2018 kern.info kernel: [101417.027053] 
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: 8021ad
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: 8021q
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: macvlan
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: veth
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: bridge
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: Network device
Tue Jul 24 16:43:13 2018 user.notice : Added device handler type: tunnel
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: wlan1-dhcp-5: interface state ENABLED->DISABLED
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: wlan1-dhcp-5: AP-DISABLED 
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: wlan1-dhcp-5: CTRL-EVENT-TERMINATING 
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan1-dhcp-5 disabled_11b_rates=0
Tue Jul 24 16:43:14 2018 kern.info kernel: [101422.572794] device wlan1-dhcp-5 left promiscuous mode
Tue Jul 24 16:43:14 2018 kern.info kernel: [101422.578305] br-dhcp: port 2(wlan1-dhcp-5) entered disabled state
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: wlan0-dhcp-2: interface state ENABLED->DISABLED
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: wlan0-dhcp-2: AP-DISABLED 
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: wlan0-dhcp-2: CTRL-EVENT-TERMINATING 
Tue Jul 24 16:43:14 2018 daemon.notice hostapd: nl80211: deinit ifname=wlan0-dhcp-2 disabled_11b_rates=0
Tue Jul 24 16:43:14 2018 kern.info kernel: [101422.604420] device wlan0-dhcp-2 left promiscuous mode
Tue Jul 24 16:43:14 2018 kern.info kernel: [101422.609865] br-dhcp: port 3(wlan0-dhcp-2) entered disabled state
Tue Jul 24 16:43:15 2018 kern.info kernel: [101423.753010] br-wan: port 2(ffuplink_wan) entered disabled state
Tue Jul 24 16:43:15 2018 kern.info kernel: [101423.775785] br-wan: port 1(eth1) entered disabled state
Tue Jul 24 16:43:15 2018 kern.info kernel: [101423.784365] device ffuplink_wan left promiscuous mode
Tue Jul 24 16:43:15 2018 kern.info kernel: [101423.789815] br-wan: port 2(ffuplink_wan) entered disabled state
Tue Jul 24 16:43:15 2018 kern.info kernel: [101423.799451] device eth1 left promiscuous mode
Tue Jul 24 16:43:15 2018 kern.info kernel: [101423.804148] br-wan: port 1(eth1) entered disabled state
Tue Jul 24 16:43:25 2018 kern.emerg kernel: [101433.952471] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
Tue Jul 24 16:43:27 2018 daemon.notice vnstatd[2835]: Interface "br-wan" disabled.
Tue Jul 24 16:43:35 2018 kern.emerg kernel: [101444.092407] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
Tue Jul 24 16:43:41 2018 daemon.err odhcp6c[1408]: Failed to send DHCPV6 message to ff02::1:2 (Permission denied)
Tue Jul 24 16:43:46 2018 kern.emerg kernel: [101454.232328] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
```
dmesg (first line repeated multiple times above and last line multipe times below)
```
[97824.593043] ath: phy0: ath_ahb_eeprom_read: eeprom data has to be provided externally
[97824.600995] ath: phy0: ath_ahb_eeprom_read: eeprom data has to be provided externally
[97833.018668] ICMPv6: process `grep' is using deprecated sysctl (syscall) net.ipv6.neigh.br-dhcp.base_reachable_time - use net.ipv6.neigh.br-dhcp.base_reachable_time_ms instead
[97833.074156] nr_pdflush_threads exported in /proc is scheduled for removal
[101417.008290] 
[101417.008290] do_page_fault(): sending SIGSEGV to netifd for invalid read access from 00000000
[101417.017039] epc = 0040c321 in netifd[400000+17000]
[101417.022040] ra  = 0040cd09 in netifd[400000+17000]
[101417.027053] 
[101422.572794] device wlan1-dhcp-5 left promiscuous mode
[101422.578305] br-dhcp: port 2(wlan1-dhcp-5) entered disabled state
[101422.604420] device wlan0-dhcp-2 left promiscuous mode
[101422.609865] br-dhcp: port 3(wlan0-dhcp-2) entered disabled state
[101423.753010] br-wan: port 2(ffuplink_wan) entered disabled state
[101423.775785] br-wan: port 1(eth1) entered disabled state
[101423.784365] device ffuplink_wan left promiscuous mode
[101423.789815] br-wan: port 2(ffuplink_wan) entered disabled state
[101423.799451] device eth1 left promiscuous mode
[101423.804148] br-wan: port 1(eth1) entered disabled state
[101433.952471] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
[101444.092407] unregister_netdevice: waiting for br-wan to become free. Usage count = 1
```

jow-, 2018-07-24:
@pmelange - thanks. I am done now, you can view the changes at the webui or by typing `iwinfo` on the cli.

Some screenshots for reference:
[image]
[image]
[image]
[image]

jow-, 2018-07-24:
Corresponding iwinfo changes at https://git.openwrt.org/?p=project/iwinfo.git;a=summary

pmelange, 2018-07-24:
Nice!

Should I leave the test setup online for any reason?

Thank you

jow-, 2018-07-24:
@pmelange - no at the moment it is not needed. I'll try to build myself a setup for testing things like joining mesh networks and properly dealing with authsae encryption for mesh.

jow-, 2020-02-23:
This should be settled in master / 19.07

closed 2020-02-23 by jow-