openwrt/luci #1520: Status page: shows wrong wan status interface (static ip configured)

avintis, 2018-01-10:
If the wan ip is static configured (not dhcp) the status page (overview) shows the wrong interface. In my case the "br-lan40" instead eth0 (see screenshot):
[image]

jow-, 2018-01-10:
This is due to the fact that you wrongly configured a gateway option on your lan interface. Remove it.

closed 2018-01-10 by jow-

avintis, 2018-01-10:
Thanks, what is wrong?

* ++ improve your information liquidity!*
*:: **Alfred Berger*, CEO
*:: *Avintis SA
*:: *route du Tir-Fédéral 10 *:: *CH-1762 Givisiez
*:: *tel: +41 26 469 71 11 *:: *fax: +41 26 469 71 10
*:: *mobil: +41 79 455 20 62
*:: *a.berger@avintis.com *:: *www.avintis.com