openwrt/luci #1522: Status page: shows wrong wan status interface (static ip configured) #1520

avintis, 2018-01-10:
reopen #1520

The answer was not solving the reported problem! Because it was not satisfactory.
 
Can you please explain what is wrong. Here my configuration.

/etc/config/network
config interface 'loopback'
        option ifname 'lo'
        option proto 'static'
        option ipaddr '127.0.0.1'
        option netmask '255.0.0.0'

config globals 'globals'
        option ula_prefix 'fdd7:ab6d:35d6::/48'

config interface 'lan'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.16.1'
        option netmask '255.255.255.0'
        option ip6assign '60'
        option _orig_ifname 'eth1.1 wlan0 wlan1'
        option _orig_bridge 'true'
        option ifname 'eth1.1'

config interface 'wan'
        option ifname 'eth0'
        option _orig_ifname 'eth0'
        option _orig_bridge 'false'
        option proto 'static'
        option ipaddr '192.168.1.120'
        option netmask '255.255.255.0'
        option gateway '192.168.1.1'
        option dns '192.168.1.1 8.8.8.8'
        option ip6assign '64'

config interface 'wan6'
        option ifname 'eth0'
        option proto 'dhcpv6'

config switch
        option name 'switch0'
        option reset '1'
        option enable_vlan '1'

config switch_vlan
        option device 'switch0'
        option vlan '1'
        option vid '1'
        option ports '0t 2 3t'

config switch_vlan
        option device 'switch0'
        option vlan '40'
        option vid '40'
        option ports '0t 3t 5'

config interface 'lan40'
        option type 'bridge'
        option proto 'static'
        option ipaddr '192.168.40.1'
        option netmask '255.255.255.0'
        option ip6assign '64'
        option gateway '192.168.40.1'
        option _orig_ifname 'eth0.40 wlan1-1'
        option _orig_bridge 'true'
        option ifname 'eth1.40'

config route
        option interface 'lan'
        option target '192.168.2.0'
        option netmask '255.255.255.0'
        option gateway '192.168.16.1'

config route
        option interface 'lan'
        option target '192.168.2.0'
        option netmask '255.255.255.0'
        option gateway '192.168.16.1'

config route
        option interface 'lan'
        option target '192.168.41.0'
        option netmask '255.255.255.0'
        option gateway '192.168.16.1'

config switch_vlan
        option device 'switch0'
        option vlan '41'
        option ports '1 6'
        option vid '2'

Thanks

jow-, 2018-01-10:
`option gateway '192.168.40.1'` in interface `lan40` is wrong. A default gateway may never point to itself, it must be empty / unset in such cases.

closed 2018-01-10 by jow-

avintis, 2018-01-10:
ok thanks solved the problem!