openwrt/luci #1535: luci-app-noddos bug

user7887, 2018-01-12:
The menu tab "Noddos Clients" is not working.
File missing, see the picture.
[image]

hnyman, 2018-01-13:
@StevenHessing 
The error might caused by no clients defined yet, and an empty file /var/lib/noddos/DeviceProfiles.json ?

It looks to me that its existence is never checked in https://github.com/openwrt/luci/commit/3b23203ce95c48a08b417a303ee8c6dc47f6eb60#diff-2ef959de0aa3fa260c2f4c92026f0cefR22

Or is the idea that the line is not reached if /var/lib/noddos/DeviceDump.json is missing, and those two files should always both exist at the same time?

StevenHessing, 2018-01-13:
The noddos package includes a cronjob that downloads a new copy of the /var/lib/noddos/DeviceProfiles.json file every 24 hours. I'll need to add a check to see if the file exists.

StevenHessing, 2018-01-14:
I can only see this happening if someone uninstalled noddos but did not uninstall luci-app-noddos. If this is indeed the use case, uninstall luci-app-noddos to mitigate the case. If noddos is still installed, please restart noddos to see if the problem disappears. The init script for noddos downloads the /var/lib/noddos/DeviceProfiles.json file.

user7887, 2018-01-14:
StevenHessing 
Well, I checked the translation. Most likely, the installation of Luci-app-Noddos (Russian language Pack) does not initiate the installation of Noddos.
Thank you! :)

closed 2018-01-14 by user7887