openwrt/luci #1540: order menu items according to dependency / proceeding when configuring

testbird, 2018-01-14:
I found that a little reordering could positively improve/guide first impression and usage.

Especially in the "Network" menu, "Switch" cold be the first and "Firewall" the third entry:

```
Network
* Switch
* Interfaces
* Firewall
* DHCP and DNS
* Hostnames
* Static Routes
* Diagnostics
```

jow-, 2018-04-06:
I agree, will reshuffle a bit

testbird, 2018-04-06:
Hi, nice to read about this!

The reason I had `Switch` at the top was because of the dependencies. For example, I needet to adjust the vlans before the `Interfaces`, because they are used in the Interface definition. Even if one has no need to adjust the switch's vlans, if I look at the `Swtich` config first, without having to change anything, I then know about the vlans and can understand the `Interfaces` page better.

testbird, 2018-04-06:
My router doesn't have wireless, so I don't know anything about the dependenies there.

testbird, 2018-04-06:
Rethinking this, I still think `Switch` would be best at the top, as it is the most basic thing configurable.

hnyman, 2018-04-06:
I think that for many/most users "switch" is never configured, as in a single home router use case there is usually no need to configure the VLANs or switch.  So, no need to have it on the top.

Wireless, interfaces, firewall and dhcp are the four top ones for me.

testbird, 2018-04-06:
Yea, it's understood that `Switch` can be left as is in most cases (which the page might even say). 

Nevertheless it was really frustrating to having to redo all the Interfaces and Firewall config configuration after I found that they depend on others. That's why i suggest to sort according to dependecies (and let the admins see the basic default settings first when they browse through the configuration interface).

jow-, 2020-07-13:
No progress for two years so it seems the status quo will prevail.

closed 2020-07-13 by jow-