openwrt/luci #1556: luci-app-noddos 2 need to fix

user7887, 2018-01-16:
1) Link together with the word:
[image]
2) Maybe just IP-addresses :):
[image]

stangri, 2018-08-10:
You will most likely need to fix this file: https://github.com/openwrt/luci/blob/master/applications/luci-app-noddos/po/ru/noddos.po and then create a Pull Request to have the correct version merged.

stangri, 2019-10-10:
There are to issues:
1. The space between localizable resource and the programmatic output is part of the localizable resource. If it's moved out of it, that should fix the problem.
2. IPv4 used as a header for the table where IPv6 addresses also present.

@feckert -- I can fix this if you want me to.

Strykar, 2020-12-21:
Upstream project is dead, this should be closed - https://github.com/noddos/noddos/issues/55#issuecomment-747105165

hnyman, 2020-12-21:
The package is to be removed.

closed 2020-12-21 by hnyman