openwrt/luci #1560: Firewall - Port forward: NAT Loopback not working for other than LAN Zone

bbogey, 2018-01-18:
I have multiple LAN zones. NAT loopback works only from LAN, not from other zones.
Please add an option to select from which zones NAT loopback should be working.

feckert, 2018-01-20:
@bbogey could you explain more where to add this?

jow-, 2018-02-13:
That would need to be implemented in OpenWrt/LEDE's firewall3 first before such an option could be added to LuCI.

closed 2018-02-13 by jow-

micw, 2020-02-09:
Hello,
it's 2 years ago since this was reported. Has the state changed? I have the same issue with 19.07.1.

GrimKriegor, 2020-02-10:
@micw Here is [the link](https://bugs.openwrt.org/index.php?do=details&task_id=1645&dev=4) for this issue in the OpenWRT bugtracker.

heapxor, 2021-11-20:
so everyone ignores such a important issue?

jmou, 2022-09-19:
For anyone else searching for a solution, this appears to have been implemented with the `reflection_zone` option in `/etc/config/firewall`. I don't think it is configurable through LuCI yet. For an example see https://github.com/openwrt/openwrt/issues/8868#issuecomment-1229441623

PR #6464 (2023-07-13): luci-app-firewall: add reflection_zone field to forwards
This allows to to define multiple zones for NAT reflection rules.

Fixes: #1560
commit 3f20598a (2023-07-13): luci-app-firewall: add reflection_zone field to forwards

This allows to to define multiple zones for NAT reflection rules.

Fixes: #1560
commit 9504081b (2023-07-27): luci-app-firewall: add reflection_zone field to forwards

This allows to to define multiple zones for NAT reflection rules.

Fixes: #1560

(cherry picked from commit 3f20598acf57759aeda8ef5448e60f5a70e532f9)
commit e4c46338 (2023-07-27): luci-app-firewall: add reflection_zone field to forwards

This allows to to define multiple zones for NAT reflection rules.

Fixes: #1560

(cherry picked from commit 3f20598acf57759aeda8ef5448e60f5a70e532f9)