openwrt/luci #1570: luci-mod-advanced-reboot - can't find 2nd partition

daloki83, 2018-01-20:
Like the title says....on Linksys WRT3200ACM
OpenWrt SNAPSHOT, r5903-a3198061f8
luci-app-advanced-reboot | git-18.020.32535-3d54..-25

It only says "Warning: This system does not have two partitions!"
Checking with "/usr/sbin/fw_printenv -n boot_part" says im on partition 2
and when i switch to partition 1 with "/usr/sbin/fw_setenv boot_part 1" and reboot, im back on Linksys OS.

Dont know how to debug this. Maybe someone can help ?

hnyman, 2018-01-20:
cc @stangri

stangri, 2018-01-21:
Thanks Hannu.

@daloki83 -- could you please run the following on the command line and post the output of:
```source /usr/share/libubox/jshn.sh; json_load "$(/bin/ubus call system board)"; json_get_vars model; echo $model```

daloki83, 2018-01-21:
Hi @stangri the output is:
"Linksys WRT3200ACM"

stangri, 2018-01-21:
My bad, fixed in https://github.com/openwrt/luci/pull/1576.

As I don't know Lua that well, that was a head scratcher.

daloki83, 2018-01-21:
Cool thank you ! - Tested and it works as expected

closed 2018-01-21 by daloki83