openwrt/luci #1599: Mount Points discrepancy

user7887, 2018-01-27:
Hello all :)
Menu 'Mount Points' is designed for mounting partitions of hard and flash drives.
Let's clarify the wording:
sda, sdb - flash or hard drive
sda1, sda2, sdb3 - flash or hard drive partitions
ext2, ext4, ntfs - flash or hard drive filesystems
/mnt/sda1, /mnt/sdb3 - mount points

Here, I propose the following changes:
https://github.com/openwrt/luci/pull/1598
See the screenshots how it will look:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Thanks!

user7887, 2018-01-27:
Now there:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

feckert, 2018-04-06:
@user7887 
From my point of view the changes make sense.
Open a PR if some one don't like it you will get a comment on this :-)

closed 2018-07-03 by jow-