openwrt/luci #1602: luci-app-wireguard: connection refused, ports not opened

neographikal, 2018-01-28:
Tried to get the wireguard extension working, installed on two LEDE routers (LEDE Reboot 17.01.4 r3560-79f57e422d) followed the guide on https://danrl.com/blog/2017/luci-proto-wireguard/ and https://www.wireguard.com/quickstart/ but the daemon does not seem to listen to the configured ports.

On both hosts:
[image]

[image]

At first I thought I made a configuration error with the keys or something, but with the ports closed it's not strange that it won't connect.

yousong, 2018-01-28:
telnet won't work here because wireguard is udp based

neographikal, 2018-01-28:
Well that was stupid, you're right ofcourse. It does start though, from syslog:

```
Sun Jan 28 12:21:14 2018 kern.info kernel: [    9.529782] nf_conntrack version 0.5.0 (1963 buckets, 7852 max)
Sun Jan 28 12:21:14 2018 kern.info kernel: [    9.633645] wireguard: WireGuard 0.0.20171017 loaded. See www.wireguard.com for information.
Sun Jan 28 12:21:14 2018 kern.info kernel: [    9.642274] wireguard: Copyright (C) 2015-2017 Jason A. Donenfeld <Jason@zx2c4.com>. All Rights Reserved.
```

neographikal, 2018-01-28:
okay that was weird, switched to a port high in the 50000's and now suddenly it connects.

closed 2018-01-28 by neographikal

yousong, 2018-01-28:
The nmap result indicates that wireguard is actually there waiting for traffic, but I also have no idea why netstat shows nothing

yousong, 2018-01-28:
Ha! it comes to my mind that port 500 is some well-known port related to ipsec.  `netstat -ap` will convert it into the named form and `grep 500` will show nothing!