openwrt/luci #1615: Brcm47xx Board Detection NOT WORKING AT ALL

tacuar, 2018-01-31:
It seems after flashing the lede firmware, the brcm47xx board detection stops working. 

Under luci first "status" page, the "model" of the wireless router shows  "unknown board" instead of the correct model number. For example, ASUS RT-N15U

When i flash back to cc15.05.1 or bb14.07, the luci will show the correct model no. of the router.

There are so many models of the brcm47xx series and all of them can not be displayed correctly

jow-, 2018-02-01:
That does not seem like a LuCI bug to me. Whats the output of `cat /etc/board.json`, `nvram show` and `cat /tmp/sysinfo/*` on the affecged devices?

jow-, 2018-02-13:
no feedback

closed 2018-02-13 by jow-

tacuar, 2018-03-01:
{
	"model": {
		"id": "0x052b:45",
		"name": "Unknown Board"
	},
	"switch": {
		"switch0": {
			"enable": true,
			"reset": true,
			"ports": [
				{
					"num": 0,
					"role": "lan"
				},
				{
					"num": 1,
					"role": "lan"
				},
				{
					"num": 2,
					"role": "lan"
				},
				{
					"num": 3,
					"role": "lan"
				},
				{
					"num": 4,
					"role": "wan"
				},
				{
					"num": 8,
					"device": "eth0",
					"need_tag": false,
					"want_untag": false
				}
			],
			"roles": [
				{
					"role": "lan",
					"ports": "0 1 2 3 8t",
					"device": "eth0.1"
				},
				{
					"role": "wan",
					"ports": "4 8t",
					"device": "eth0.2"
				}
			]
		}
	},
	"network": {
		"lan": {
			"ifname": "eth0.1",
			"protocol": "static"
		},
		"wan": {
			"ifname": "eth0.2",
			"protocol": "dhcp"
		}
	}
}

tacuar, 2018-03-01:
FLASHED AGAIN AND IT'S STILL THE SAME. PROBLEM EXISTS BOTH ON LEDE17014 AND LATEST LEDE SNAPSHOT

tacuar, 2018-03-01:
i checked in google and i noticed some history commits record of this patch for  brcm47xx.

in one commits it changed asus brcm47xx from RTN12HP to RT-N12HP, it deleles RTN12HP and add RT-N12HP, but that is the openwrt commits.

I'm not sure if this could be the reason that the lede does not recognize the ASUS brcm47xx boards.

It detects my RT-N15U asus router as a UNKNOWN BOARD.

jow-, 2018-03-01:
1) this is no LuCI issue at all
2) you still didn't provide the output of `nvram show`

tacuar, 2018-03-01:
ok i will send the nvram info

tacuar, 2018-03-01:
@jow- 

root@LEDE:~# nvram show
boardnum=45
sb/1/antswitch=0x0
landevs=vlan1 wl0
sb/1/pa2gw0a0=0xFEFD
hardware_version=RTN15U-1.0.1.3
sb/1/pa2gw0a1=0xFEFD
sb/1/cck2gpo=0x0
sdram_ncdl=0x00000000
sb/1/macaddr=50:46:5D:0B:32:48
sb/1/boardflags=0x80001710
sb/1/boardflags2=0x1000
sb/1/pa2gw1a0=0x1451
sb/1/pa2gw1a1=0x1503
sb/1/stbcpo=0x0
boardpwrctl=0xd00
wl_msglevel=0x1
sb/1/pa2gw2a0=0xFAFB
sb/1/pa2gw2a1=0xFADC
sb/1/cddpo=0x0
sb/1/pdetrange2g=0x2
sb/1/bw40po=0x0
sb/1/devid=0x4329
sb/1/ccode=XU
sdram_config=0x106
vlan1hwname=et0
sb/1/leddc=0xffff
sb/1/regrev=0
lan_netmask=255.255.255.0
sb/1/sromrev=8
vlan2ports=4 8
reset_gpio=8
et0macaddr=50:46:5D:0B:32:48
sb/1/elna2g=2
antswctl2g=0x1
sb/1/ofdm2gpo=0x66554433
regulation_domain=XU
sdram_init=0x0419
sb/1/tssipos2g=0x1
et0phyaddr=30
boardflags=0x80001710
wandevs=vlan2
sb/1/temps_hysteresis=5
lan_ipaddr=192.168.1.1
clkfreq=500,200,100
wait_time=3
sb/1/bwduppo=0x0
sb/1/aa2g=0x3
et_swleds=0x0
boardtype=0x052b
sb/1/triso2g=0x3
pmon_ver=CFE 5.60.127.30 @VERSION_TYPE@
sb/1/mcs2gpo7=0xAA99
sb/1/mcs2gpo3=0x6655
sb/1/mcs2gpo4=0x8877
sb/1/mcs2gpo5=0xAA99
boot_wait=on
sb/1/temps_period=5
sb/1/mcs2gpo6=0x8877
sb/1/mcs2gpo0=0x4433
sb/1/mcs2gpo1=0x6655
sb/1/mcs2gpo2=0x4433
boardrev=0x1204
opo=0x0
sb/1/rxchain=0x3
sb/1/ledbh3=11
sb/1/ledbh0=8
sb/1/maxp2ga0=0x50
sb/1/ledbh1=3
sb/1/maxp2ga1=0x50
sb/1/ledbh2=11
sb/1/antswctl2g=0x1
sdram_refresh=0x8040
vlan1ports=0 1 2 3 8*
secret_code=62935784
boardflags2=0x1000
sb/1/txchain=0x3
xtalfreq=20000
sb/1/extpagain2g=0x2
vlan2hwname=et0
sb/1/ag0=0x2
sb/1/ag1=0x2
sb/1/itt2ga1=0x20
sb/1/tempthresh=120
sb/1/itt2ga0=0x20
gpio7=robo_reset
et0mdcport=0
watchdog=0

tacuar, 2018-03-01:
hardware_version=RTN15U-1.0.1.3

this router is ASUS RT-N15U, belongs to BRCM47XX series i suppose.

Not sure if its due to this model is not listed in LEDE brcm47xx board detection range? at this moment when i flash lede17014 or lede most latest snopshot version, it's deteced as a unknown board in LUCI.

tacuar, 2018-03-01:
the CFE is the original CFE, for ASUS RT-N15U model. 

It has not been modified any.

jow-, 2018-03-01:
This needs to be fixed in OpenWrt and the upstream kernel.
It seems that Linux [looks for](https://elixir.bootlin.com/linux/v4.9.85/source/arch/mips/bcm47xx/board.c#L247) `productid=RT-N15U` in nvram to identify the model, your dump above does not contain that.

OpenWrt userspace model detection relies on the kernel being able to identify the board. Either your nvram got corrupted somehow (certain models do not restore proepr defaults after nvram erase) or the upstream Kernel support was added based on wrong assumptions.

In either case this is nothing which can be fixed here, please report it at https://bugs.openwrt.org/.
In the meanwhile you should be able to workaround the problem with `nvram set productid=RT-N15U; nvram commit; reboot`

tacuar, 2018-03-02:
@jow-  So professional of you Jow! Personally i think it is the second issue that you listed above, the upstream Kernel support was added based on wrong assumptions.

And plus, thank you for the solution of nvram set, which i once used in dd-wrt. I will try it on this openwrt firmware and see if it works.

tacuar, 2018-03-02:
the command does not work either:   
nvram set productid=RT-N15U
nvram commit
reboot

and also it seems the openwrt / lede does not support very well of this ASUS RT-N15U:
when the flash of the firmware to this router is completed, the first boot is ok, under 192.168.1.1, the setup and configuration goes on well.

Then if the router is swithed off and , if the power is on again, it stays at  TTL=100 all the time and can not go into the lede / openwrt system. 

That means, after burning the openwrt / lede, and if i reboot the router, it will automatically go into TTL=100 status and , if i type 192.168.1.1, it goes into the CFE interface.

If i flash back to ASUS factory firmware, then it does not have this problem.

dinn942021, 2021-06-27:
> the command does not work either:
> nvram set productid=RT-N15U
> nvram commit
> reboot
>
> and also it seems the openwrt / lede does not support very well of this ASUS RT-N15U:
> when the flash of the [...]

dinn942021, 2021-06-27:
hi, year 2021. have you find any solution to this?

mirh, 2024-05-16:
d7d10f2c1e8511fe07c9760e85f2272a85168f8d or 9d95d5e799a3fd2c472cafe7ca5cc987d7183e69 may have helped?