openwrt/luci #1628: lots of error in wifi page and mount point

Ansuel, 2018-02-13:
router: r7800
openwrt trunk 
(gcc 7.3)

i got this error when i try to modify the wifi settings
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: /usr/lib/lua/luci/dispatcher.lua:487: Failed to execute arcombine dispatcher target for entry '/admin/network/wireless/radio0.network1'.
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: The called action terminated with an exception:
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: /usr/lib/lua/luci/model/cbi/admin_network/wifi.lua:216: bad argument #3 to '?' (integer expected, got number)
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: stack traceback:
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: 	[C]: in function 'assert'
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: 	/usr/lib/lua/luci/dispatcher.lua:487: in function 'dispatch'
Tue Feb 13 11:53:55 2018 daemon.err uhttpd[669]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: /usr/lib/lua/luci/dispatcher.lua:487: Failed to execute arcombine dispatcher target for entry '/admin/network/wireless/radio1.network1'.
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: The called action terminated with an exception:
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: /usr/lib/lua/luci/model/cbi/admin_network/wifi.lua:216: bad argument #3 to '?' (integer expected, got number)
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: stack traceback:
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: 	[C]: in function 'assert'
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: 	/usr/lib/lua/luci/dispatcher.lua:487: in function 'dispatch'
Tue Feb 13 11:54:09 2018 daemon.err uhttpd[669]: 	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

and 500 error in the mount point page
/usr/lib/lua/luci/dispatcher.lua:487: Failed to execute cbi dispatcher target for entry '/admin/system/fstab'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/tblsection'.
A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/dvalue'.
A runtime error occured: /usr/lib/lua/luci/model/cbi/admin_system/fstab.lua:160: bad argument #4 to '?' (integer expected, got number)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:487: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function

jow-, 2018-04-06:
Can you please provide the output of `lua -lluci.util -liwinfo -e 'luci.util.dumptable(iwinfo.nl80211.txpwrlist("radio0"))'` if you stell encounter this error?

Ansuel, 2018-04-20:
@jow- 
here
`root@No-More-Lag-Router:~# lua -lluci.util -liwinfo -e 'luci.util.dumptable(iwinfo.nl80211.txpwrlist("radio0"))'
1       table: 0x6a3530
        dbm     0
        mw      1
2       table: 0x6aba30
        dbm     1
        mw      1
3       table: 0x6aba60
        dbm     2
        mw      1
4       table: 0x6abae0
        dbm     3
        mw      1
5       table: 0x6aa390
        dbm     4
        mw      2
6       table: 0x6abc00
        dbm     5
        mw      3
7       table: 0x68f8b0
        dbm     6
        mw      3
8       table: 0x6a7280
        dbm     7
        mw      5
9       table: 0x6a5740
        dbm     8
        mw      6
10      table: 0x6a16c0
        dbm     9
        mw      7
11      table: 0x697bb0
        dbm     10
        mw      10
12      table: 0x6952b0
        dbm     11
        mw      12
13      table: 0x6a5dd0
        dbm     12
        mw      15
14      table: 0x69e8a0
        dbm     13
        mw      19
15      table: 0x6a57f0
        dbm     14
        mw      25
16      table: 0x6a2f90
        dbm     15
        mw      31
17      table: 0x694fc0
        dbm     16
        mw      39
18      table: 0x6a5970
        dbm     17
        mw      50
19      table: 0x6a59f0
        dbm     18
        mw      63
20      table: 0x6a6f70
        dbm     19
        mw      79
21      table: 0x6a4330
        dbm     20
        mw      100`

Ansuel, 2018-04-23:
i still have this problem.

jow-, 2018-04-24:
Yes, unfortunately I am still unable to reproduce the problem. Can you please edit `/usr/lib/lua/luci/model/cbi/admin_network/wifi.lua`, find the line `tp:value(p.driver_dbm, "%i dBm (%i mW)"` and add the following, new line before it:

    io.stderr:write("dbm=%s mw=%s\n" %{ tostring(p.display_dbm), tostring(p.display_mw) })

Then execute `rm -r /tmp/luci-*` and reproduce the error, finally check `logread | grep dbm=` and post the resulting lines here.

Ansuel, 2018-04-24:
`root@No-More-Lag-Router:~# logread | grep dbm=
Tue Apr 24 15:08:11 2018 daemon.err uhttpd[2896]: dbm=0 mw=1
Tue Apr 24 15:08:11 2018 daemon.err uhttpd[2896]: dbm=1 mw=1.2589254117942`

(also i would like to report a problem with the space in the device...) 

Spazio libero: **67.364693446089%** (2.51 MB)     (think having a float for a % is wrong...)

Ansuel, 2018-04-25:
@jow- ok looks like math.floor is broken.... it should convert 95,25 to 95 but it doesn't do it...  any idea why ?

Ansuel, 2018-04-25:
@jow- prepared a pull request to fix this... hope it helps... 1768

closed 2018-05-23 by Ansuel