openwrt/luci #1636: Unable to configure multiple IPv6 prefixes in luci

grandiego, 2018-02-21:
I’m running OpenWrt SNAPSHOT r6074-267873a / LuCI Master (git-18.039.58622-76f9f5e) on Alpha ASL26555 (16M). Tried also in 15.05.1, same behavior.

I’m running a 6in4 tunnel for IPv6 connectivity, provided by Hurricane Electric.

HE delegates to me two IPv6 prefixes, a /64 and a /48.
I’ve set up the tunnel in lucy effortlessly, but just the /48 was working.

After pulling my hair for a long time, I figured out to add by hand another ip6prefix line to my tunnel interface in /etc/config/network, but only the last ip6prefix sentence is honored.

In lucy, There’s a field named ‘IPv6 routed prefix’ that only allows exactly one prefix (and is the last one listed in /etc/config/network interface declaration). The field should be ‘extensible’, so multiple prefixes could be added, much like ‘forwarding DNS’ settings in Network/DHCP and DNS settings in lucy.

I'm reporting here as suggested in https://bugs.lede-project.org/index.php?do=details&task_id=1361

jow-, 2018-10-20:
This has been fixed a while back.

closed 2018-10-20 by jow-

mikma, 2018-11-28:
> This has been fixed a while back. 

@jow- May I ask how you configure multiple ip6prefix entries on one interface in luci?

I'm using a recent snapshot release, see below, but I can't figure out how to do it. Netifd does support multiple prefixes in ip6prefix. Both as a genuine list and as a string using spaces as separators. But on the Luci "interfaces" page at most one IPv6-PD is shown for each interface. The same is true for "IPv6 routed prefix" on the interface page. When you try to enter multiple prefixes separated with spaces (or colons) then the field turns red.

```
DISTRIB_ID='OpenWrt'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r8576-731a12a'
DISTRIB_TARGET='x86/64'
DISTRIB_ARCH='x86_64'
DISTRIB_DESCRIPTION='OpenWrt SNAPSHOT r8576-731a12a'
DISTRIB_TAINTS=''

```

jow-, 2018-11-28:
This report was for 6to4, 6in4 and the other IPv6 tunnel protocols. You likely try to configure a proto static interface? In this case the same datatype validation fix needs to be applied there. Reference: https://github.com/openwrt/luci/commit/60e7ea2fc8363960861a82f1e65ca4a7a1d60bfe