openwrt/luci #1637: luci-app-firewall: cannot use multiple ports for the same rule

luizluca, 2018-02-21:
On LEDE 17.01.4:

UCI and fw3 accepts rules like:

```
config rule
        option enabled '1'
        option target 'ACCEPT'
        option proto 'tcp'
        option name 'adm open'
        option src 'lan'
        option src_ip '10.1.1.0/24'
        list dest_port '80'
        list dest_port '22'
```

LuCI even already shows it nicely in "Traffic rules" tables as "To any router IP at ports 80, 22 on this device"

However, the "Edit Rule" view only allows one port. It simply shows the first port (in case, 80). If I change port 80 to 81, LuCI changes the port entries from "list" to "option", remove all existing port references and use only the new port (81).

luizluca, 2018-02-21:
The same goes for src_ip.

EricLuehrsen, 2018-05-07:
Either "_port" option can be entered as a space separated lists. That is one work around at least. The "_ip" options do not work this way though.

jow-, 2018-05-07:
Will take care of it soon (tm) - I promise. The main hurdle here is the lack of suitable widgets. The rule detail forms are a huge mess already and I need to think about a way to structure them better.

commit 5142e40f (2018-06-22): luci-app-firewall: update cbi models

 - allow multiple src/dest ips for rules (#1637)
 - restrict ICMP type list to ICMP protocol
 - add section title callbacks
 - remove size annotations
 - fix validation error with aliased zone fields (#1882)

Fixes #1637, #1882.
closed 2018-06-22 by jow-