openwrt/luci #1646: Show the architecture in the overview page

jeroenh, 2018-02-27:
Updating my OpenWRT is not something I do regularly. Every time I do, I go to the OpenWRT download website and I'm faced with a huge list of architectures. Every time, I forget which architecture I should use, and I have to search the wiki to hopefully find that simple piece of information.

It would be a huge timesaver if Luci could show me the architecture of my firmware directly.
Or even better yet, link directly to the latest firmware for my device.

dibdot, 2018-02-27:
In Luci you'll see a subset of ...
```
ubus call system board
{
	"kernel": "4.9.82",
	"hostname": "adb2go",
	"system": "Qualcomm Atheros QCA9533 ver 2 rev 0",
	"model": "GL.iNet GL-AR750",
	"board_name": "gl-ar750",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r6222+89-e1d4aa36b0",
		"target": "ar71xx\/generic",
		"description": "OpenWrt SNAPSHOT r6222+89-e1d4aa36b0"
	}
}

```

jeroenh, 2018-02-27:
The target field is what I would want to see, but it's not shown on the web interface.

dibdot, 2018-02-27:
It was meant as a workaround for you to get the information in a normal ssh session. I doubt that the normal user is interested in target / architecture information.

jeroenh, 2018-02-27:
Surely you want users of OpenWRT to stay up to date and make it as easy as possible for them to upgrade?

dibdot, 2018-02-27:
Normal users download release tarballs via OpenWrt website, look here: https://openwrt.org/toh/views/toh_fwdownload

dibdot, 2018-02-27:
And maybe you check "luci-app-attendedsysupgrade" if this app fulfill your needs.

dibdot, 2018-04-07:
Possible workarounds are illustrated, so I am going to close this one.

closed 2018-04-07 by dibdot