openwrt/luci #1649: A runtime error occured: /usr/lib/lua/luci/model/network.lua:187: attempt to index a nil value

fnarfbargle, 2018-02-28:
I'm using a recent trunk self-build on a WRT32x modified to think it is a WRT3200. 
Linux WRT32x 4.14.18 #0 SMP Tue Feb 13 22:21:12 2018 armv7l GNU/Linux
Current tree head : 
commit f5b4f5f8e33624f27af9fb3f86e09084181c08ed
Date:   Sun Feb 25 03:18:41 2018 +0000

I see the following in the logs once or twice a day.

Feb 27 22:48:25 WRT32x uhttpd[1924]: /usr/lib/lua/luci/dispatcher.lua:487: Failed to execute template dispatcher target for entry '/admin/status/overview'.
Feb 27 22:48:25 WRT32x uhttpd[1924]: The called action terminated with an exception:
Feb 27 22:48:25 WRT32x uhttpd[1924]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'admin_status/index'.
Feb 27 22:48:25 WRT32x uhttpd[1924]: A runtime error occured: /usr/lib/lua/luci/model/network.lua:187: attempt to index a nil value
Feb 27 22:48:25 WRT32x uhttpd[1924]: stack traceback:
Feb 27 22:48:25 WRT32x uhttpd[1924]: #011[C]: in function 'assert'
Feb 27 22:48:25 WRT32x uhttpd[1924]: #011/usr/lib/lua/luci/dispatcher.lua:487: in function 'dispatch'
Feb 27 22:48:25 WRT32x uhttpd[1924]: #011/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

The line in question is : 
local phy_only = force_phy_only or (ipc.link(ifname).type ~= 1)

How would I go about debugging this?

Regards,
Brad

jow-, 2018-02-28:
How's the memory utilization on the device?

fnarfbargle, 2018-02-28:
Pretty solid at about 53%. The device is operating as an AP only. No firewall, no routing, nothing fancy. It just shunts packets over the bridge.

fnarfbargle, 2018-04-11:
Nope. Recompiled from git head last night, checked relevant fix was in the tree. It happened again at 6 this morning.

network.interface dump attached.

[network.interface.dump.txt](https://github.com/openwrt/luci/files/1897666/network.interface.dump.txt)