openwrt/luci #1657: Luci Bug on Trunk /usr/lib/lua/luci/model/network.lua:1005

linuxonlinehelp, 2018-03-02:
Update today cause ath9k drop issues after 12hours:

System:
OpenWrt SNAPSHOT r6349-9046e92 / LuCI Master (git-18.060.68662-3e17668) 

Log:

Fri Mar  2 14:19:50 2018 daemon.err uhttpd[2409]: The called action terminated with an exception:
Fri Mar  2 14:19:50 2018 daemon.err uhttpd[2409]: /usr/lib/lua/luci/model/network.lua:1005: bad argument #2 to '?' (no value)
Fri Mar  2 14:19:50 2018 daemon.err uhttpd[2409]: stack traceback:
Fri Mar  2 14:19:50 2018 daemon.err uhttpd[2409]:       [C]: in function 'assert'
Fri Mar  2 14:19:50 2018 daemon.err uhttpd[2409]:       /usr/lib/lua/luci/dispatcher.lua:487: in function 'dispatch'
Fri Mar  2 14:19:50 2018 daemon.err uhttpd[2409]:       /usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

Error seen on NETWORK-> INTERFACES  Echo "Collecting ..."

Howto fix?
Thanks

jow-, 2018-03-02:
Please provide the following SSH command output:
`ubus call network.interface dump`

jow-, 2018-03-02:
Nevermind, fix pushed.

closed 2018-03-02 by jow-

linuxonlinehelp, 2018-03-02:
Fri Mar  2 14:53:53 2018 daemon.err uhttpd[2548]: The called action terminated with an exception:
Fri Mar  2 14:53:53 2018 daemon.err uhttpd[2548]: /usr/lib/lua/luci/model/network.lua:1005: bad argument #2 to '?' (no value)
Fri Mar  2 14:53:53 2018 daemon.err uhttpd[2548]: stack traceback:
Fri Mar  2 14:53:53 2018 daemon.err uhttpd[2548]:       [C]: in function 'assert'
Fri Mar  2 14:53:53 2018 daemon.err uhttpd[2548]:       /usr/lib/lua/luci/dispatcher.lua:487: in function 'dispatch'
Fri Mar  2 14:53:53 2018 daemon.err uhttpd[2548]:       /usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

burned latested sysupgrade .. same
OpenWrt SNAPSHOT r6350-175b262328 / LuCI Master (git-18.060.68662-3e17668) 

my mistake?

commit bf49505e (2018-03-02): luci-base: properly handle undefined IPv6 local-address information

If IPv6 prefix assignment is disabled, the "local-address" structure
might exist, but be empty which causes the adress formatting in the
network model class to bail out.

Verify the completeness of the "local-address" structure before using
it in order to avoid runtime errors.

Fixes #1657.