openwrt/luci #1659: Expose the isolate wireless option in luci

jonathanunderwood, 2018-03-03:
At present, Luci doesn't seem to show an option to set the configuration setting "isolate" for wireless networks. For example, when setting up a guest wireless network, it's often helpful to prevent devices on that network from communicating with each other. [Such an option is available in `/etc/config/wireless`](https://wiki.openwrt.org/doc/recipes/guest-wlan), but doesn't seem to be exposed in luci.

guidosarducci, 2018-03-22:
@jonathanunderwood I think this is something many people have wanted to see in the Luci interface for some time. Have you looked into this enough to determine if there are any technical or implementation caveats?

dibdot, 2018-03-26:
The "isolate" option has been added with #1140, it's part of trunk & current release 17.01.

[image]

closed 2018-03-26 by dibdot

jonathanunderwood, 2018-03-26:
Wow, Excellent. Many thanks for the work on this.

guidosarducci, 2018-03-26:
@dibdot Could you please double-check?

I can see that this same patch has *not* been applied to the `lede-17.01` branch, and the "isolate" option is still missing under the `hwtype == "mac80211"` UI definitions.

Hopefully just an oversight? Can this low-risk change be back-ported then?

dibdot, 2018-03-26:
Apologize, it was too early in the morning ... it's only implemented for atheros/broadcom in 17.01 and not for mac80211. So please take a current snapshot or wait for the next stable release 18.x which will be probably branched in april/may.

guidosarducci, 2018-03-26:
Ah, no problem, I just got excited thinking this was done! Would you object to my testing the change on lede-17.01 and making a back-port PR?  Thanks...