openwrt/luci #1661: DHCP crash beacause insufficient regex checks

RuderLaplace, 2018-03-04:
In firm 15.363.78009-956be55, the web form allows to enter a reserved static ip hostname with a character "_" at the begining of the name. Tought whenever it restart the service crashes due to this.

dibdot, 2018-03-05:
15.363 looks like an ancient, home brewed chaos calmer release ... did you retest with a current snapshot or latest stable release 17.01.x series?

RuderLaplace, 2018-03-17:
17.01.4 r3560-79f57e422d
Same problem.
Sun Mar 18 00:33:31 2018 daemon.crit dnsmasq[1098]: bad DHCP host name at line 27 of /var/etc/dnsmasq.conf.cfg02411c
Sun Mar 18 00:33:31 2018 daemon.crit dnsmasq[1098]: FAILED to start up

dibdot, 2018-03-18:
@RuderLaplace Thanks for re-testing!

I can confirm that this still exists in current trunk. Root cause is the used "hostname" dataytpe for validation, which allows an underscore at the beginning.

@jow- @hnyman Should we add an additional datatype "dhcphost" which does not allow an "_" at the beginning or should we fix this only in dhcp.lua or maybe a totally different approach?

RuderLaplace, 2018-03-18:
Doing a bit of research, I found out the RCF syntaxis:
* grep regexp:  "^[a-zA-Z]\([a-zA-Z0-9\-]*[a-zA-Z0-9]\)*$"
* Description: «[...] They must start with a letter, end with a letter or digit, and have as interior characters only letters, digits, and hyphen.
 [...] Labels must be 63 characters or less.»
* Syntaxis:
<domain> ::= <subdomain> | " "
<subdomain> ::= <label> | <subdomain> "." <label>
<label> ::= <letter> [ [ <ldh-str> ] <let-dig> ]
<ldh-str> ::= <let-dig-hyp> | <let-dig-hyp> <ldh-str>
<let-dig-hyp> ::= <let-dig> | "-"
<let-dig> ::= <letter> | <digit>
<letter> ::= any one of the 52 alphabetic characters A through Z in upper case and a through z in lower case
<digit> ::= any one of the ten digits 0 through 9

*Ref: https://tools.ietf.org/html/rfc1034#section-3.5

You're welcome ;)

dibdot, 2018-04-07:
Fixed with 94ea907 and 70ffbe6

closed 2018-04-07 by dibdot