openwrt/luci #1663: OpenVPN build with --enable-pkcs11 fails

Telefonorosso, 2018-03-05:
## Problem Description
On Lede image builder under Debian i686 with 4.9.0 kernel, if I edit package/network/services/openvpn/Makefile and change

> --disable-pkcs11 to --enable-pkcs11

build fails. The configure script cannot find pkcs11-helper dependency.

## Proposed Resolution
Add the pkcs11-helper software in Lede

## Steps to reproduce
nano [BUILD_DIR]/package/network/services/openvpn/Makefile
make package/openvpn/compile

hnyman, 2018-03-05:
Wrong repo.
That openvpn package is in the main Openwrt/LEDE sources, not here in the LuCI repo for GUI.

closed 2018-03-05 by hnyman

Telefonorosso, 2018-03-05:
Ok thanks.
Can you provide me the link?
THX

hnyman, 2018-03-05:
https://bugs.openwrt.org