openwrt/luci #1667: luci-theme-material Firewall label.zonebadge misaligned

zakaszewski, 2018-03-07:
Hi all,

first of all I hope that this is the right place to raise an issue regarding the material theme, if not, please let me know where would be appropriate.

I've noticed a little "odd" behaviour of the theme whenever firewall zones are displayed in a selection list:

[image]

As You can see the alignment of the "zone background" is slightly off, which appears in both
- network > firewall > [any rule] / [any port forward] and
- network > interfaces > [any interface] > firewall settings.

Other themes (e.g. bootstrap) do not seem to have the same issue, as far as I know.

By the way, I've seen the same behavior on all browsers:
Chrome 65.0.3325.146
Firefox 58.0.2
Internet Explorer 11.0.9600.18920

I know it's a minor issue but it would be much appreciated if it could be fixed.

Best regards,

Piotr

dibdot, 2018-03-08:
Did you mean like that?

[image]
[image]

zakaszewski, 2018-03-09:
That would be consistent with how Firewall General settings looks :-)

dibdot, 2018-03-09:
Patch applied, should be fixed now in trunk.

closed 2018-03-09 by dibdot