openwrt/luci #1670: Realtime Graph - Wireless Tab

kione, 2018-03-07:
It seems that the build LuCI Master (git-18.061.77516-2913111) in Real Time Graphs - Wireless doesn't work. The Traffic view is performing very fine.
[image]

jow-, 2018-03-07:
What is the actual problem? I can’t infer any information from the screenshot you provided.

kione, 2018-03-08:
I don’t see an real time data. In my screenshot I tried to show a iper3 output which was made seconds ago and the graph from the lucy webinterface. Affecting the described tab only

jow-, 2018-03-08:
Sorry I still don't understand. The graph shows data?

kione, 2018-03-08:
Sure it shows data. But not the actual rate. It shows a static Average of 39Mbit, despite the rate should be 253Mbit. See this in my screenshot. The black window is a terminal, iperfing to my lede router. As i sayed previously, the traffic tab wlan0 is correct on this. Sorry to not be more clear

jow-, 2018-03-08:
The graph displays phy rate, not throughput. It also shows an aggregated average, so if multiple stations are connected it will render the average phy rate of all stations. The chart can only be meaningfully correlated to any kind of iperf results if the router is in client mode or if only one single station is connected.

kione, 2018-03-08:
In short, I don't get the point of this view. Since it shows Mbit I can't  calculate a senseful messure. I did not want to bother you. Thanks for taking care.

closed 2018-03-08 by kione