openwrt/luci #1673: Presence of wifi-iface "monitor" removes channel information from wireless page

jeffsf, 2018-03-09:
If a monitor wifi-iface is created, the /luci/admin/network/wireless no longer shows the channel for the radio.

With
```
config wifi-iface
	option device 'radio0'
	option mode 'monitor'
	option ifname 'mon24'
```
[image]

Without
[image]

jeffsf, 2018-03-09:
This is potentially due to or related to what `iw dev` returns when configured with a monitor interface
```
phy#1
[...]
	Interface wlan1-1
		ifindex 29
		wdev 0x100000003
		addr <redacted>
		ssid <redacted>
		type AP
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 25.00 dBm
	Interface mon24
		ifindex 28
		wdev 0x100000002
		addr <redacted>
		type monitor
		txpower 25.00 dBm
phy#0
	Interface wlan0
[...]
```
There is no "wlan1" and if the monitor interface had not been renamed, it still would have carried no channel information.

Archer C7, LEDE 17.01 branch, Powered by LuCI lede-17.01 branch (git-18.039.58469-1c94003)

jow-, 2018-04-05:
This is already fixed in master.

closed 2018-04-05 by jow-