openwrt/luci #1674: "Associated Stations" - regex for parsing /etc/ethers fails for valid files

jeffsf, 2018-03-09:
As written, LuCI fails to parse valid  `/etc/ethers` files. 

As a result, the "Associated Stations" section does not reflect the contents of `/etc/ethers`

As per `ethers(5)`

> The  two items are separated by any number of SPACE and/or TAB characters.  A # at the beginning of a line starts a comment which extends to the end of the line.  The Ethernet-address is written as `x:x:x:x:x:x`, where x is a hexadecimal number between 0 and  ff  which  represents one byte of the address, which is in network byte order (big-endian).  The IP-number may be a hostname which can be resolved by DNS or a dot separated number.

In `_nethints(what, callback)` in `lua/luci/sys.lua`

```
	if fs.access("/etc/ethers") then
		for e in io.lines("/etc/ethers") do
			mac, ip = e:match("^([a-f0-9]%S+) (%S+)")
			if mac and ip then
				_add(what, mac:upper(), ip, nil, nil)
			end
		end
	end
```

While I'm not a LUA expert, that regex is looking for a single space between two non-space characters. I have confirmed this by testing with `/etc/ethers` and, for example, a tab character does not result in the UI picking up the second field, while a single space does. It also appears to require lower-case characters, which it then immediately calls upper() on for display in the UI. You can't, for example, copy from the UI and paste into `/etc/ethers`

I believe that regex could be better written as 
```
^([a-fA-F0-9:]+)%s+(%S+)
```
(a LUA expert should check this)

Yes, the `man` page doesn't specify that there can't be spaces at the start of the line, but that a tab or a series of spaces wasn't accepted was very surprising and resolving that would be greatly appreciated.

yousong, 2018-03-09:
Hi @jeffsf , Thanks for reporting to us.  Very likely you are right according to the manual, but I am also interested how the current string pattern causes trouble.

To be honest, personally I prefer these format to be strict to ease both implementation and usage, to the extent to keep it the current way in order to enforce this strictness ;)

dibdot, 2018-03-09:
@yousong sorry, saw your comment to late - will close my PR later

jeffsf, 2018-03-09:
@yousong 
First off, not all OpenWRT/LEDE boxes that have wireless enabled and run LuCI are running `dnsmasq` and/or `odhcpd`. This can  be due to multiple APs and a single DHCP server, as well as use of DHCP services supplied elsewhere in the network. As a result, there is no DHCP leases file on the OpenWRT/LEDE device to supply MAC-to-name translation.

Now as to the formatting, since `/etc/ethers` is a common construct, from `libc` up through both the Linux and BSD world, so these files may and are maintained by tools or individuals that expect compliance with those reasonable standards. As an example, these may be databases or spreadsheets where it is very common practice to use the tab character as a field delimiter, if not by default.

```
     The function ether_line() scans l, an ASCII string in ethers(5) format and sets e to the ethernet address specified in the
     string and h to the hostname.  This function is used to parse lines from /etc/ethers into their component parts.
```

If you personally _choose_ to create your `/etc/ethers` with a single space, I have no issues with that.

For me, compliance with widely accepted standards for the `/etc/ethers` file and the generally wise decision to not programmatically distinguish between spaces and tabs in human-readable and -editable files, _that does not impinge on your decision to use a single space_ is a much better decision.

@dibdot -- Thanks for the quick response. Please keep the PR open.

jeffsf, 2018-03-09:
Note also that tab is the default for the common `cut` tool
```
CUT(1)                                                            User Commands                                                           CUT(1)

NAME
       cut - remove sections from each line of files

SYNOPSIS
       cut OPTION... [FILE]...

DESCRIPTION
       Print selected parts of lines from each FILE to standard output.

       With no FILE, or when FILE is -, read standard input.

       Mandatory arguments to long options are mandatory for short options too.

       -b, --bytes=LIST
              select only these bytes

       -c, --characters=LIST
              select only these characters

       -d, --delimiter=DELIM
              use DELIM instead of TAB for field delimiter
```

yousong, 2018-03-10:
Hi @jeffsf, again, you are very likely right according these manuals.   Sure, it's just tiny bits of change to luci and it surely do not hurt so I am totally okay with the proposed change.

But the same also holds that I do not care very much in this case.  It works now and will continue to work in current settings.  I just do not bother the change.  That's why I am interested in the actual situation that the current string pattern is actually cause trouble.

As for cut, paste using tabs as the default sep.  Well, people will need to use `-d' \n'` now ;)

jeffsf, 2018-03-10:
I'm glad you're totally OK with the proposed change.

I described it in detail above; providing data on MAC-to-name translation for the LuCI UI or potentially code that uses the LuCI routines for systems that have wireless clients attaching that do not have corresponding DHCP lease entries on the specific device on which LuCI is running.

commit 62630fbf (2018-03-11): luci-base: fix parsing of ethers(5)

The /etc/ethers file may contain any number of white space characters
between the mac address and the IP/hostname field, so extend the pattern
to allow for that.

Man ethers(5) also states that the IP field may be a symbolic hostname,
so test whether the name is an IP address or hostname before adding it
to the hints structure.

Fixes #1674.
closed 2018-03-11 by jow-