openwrt/luci #1700: Auth token with Null username and password

GabrieleBazzottiPers, 2018-03-26:
When I try to log in with
`curl https://<hostname>/cgi-bin/luci/rpc/auth --data '{ "id":1, "method":"login", "params" : [null,null] }'`
I get `{"id":1,"result":"a3aa53c842f6ec306e1480a4c7cef5b1","error":null}`
The token don't let me do command like get_all and set but it should't  exist in the response.

sophana, 2018-04-01:
in sys.lua, user.checkpasswd looks really suspicious. It returns true when pwh == nil???
A simple fix would be to check username, but I think there is a big (and severe) problem there.
```
function user.checkpasswd(username, pass)
	local pwh, pwe = user.getpasswd(username)
	if pwe then
		return (pwh == nil or nixio.crypt(pass, pwh) == pwh)
	end
	return false
end
```

jow-, 2018-04-05:
> in sys.lua, user.checkpasswd looks really suspicious. It returns true when pwh == nil???

When pwh is nil, no password is set in `/etc/shadow`, in this case access is allowed regardless of the provided password.

jow-, 2018-04-06:
Fixed with feefc60 - `null, null` authentication now returns `null` instead of a defunct token

closed 2018-04-06 by jow-